FreeCAD/FreeCAD #18784: Freecadcmd

oktabruna, 2024-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

the conversion to an STL file with the freecadcmd version OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39153 (Git)
works.

but not with the latest

this is the script

import os
import sys
import argparse

import math

import FreeCAD
import FreeCADGui
import Part
import Mesh
import MeshPart

our_args = sys.argv[-4:]

parser = argparse.ArgumentParser()
parser.add_argument('--file', nargs='+')
parser.add_argument('--stl', nargs='+')

parsed = parser.parse_args(our_args)

if len(parsed.file) > 1 or len(parsed.stl) > 1:
    print("Too many arguments for either --file or --stl")
in_fn = os.path.normpath(parsed.file[0])
out_fn = os.path.normpath(parsed.stl[0])

FreeCADGui.setupWithoutGUI()
native_file = FreeCAD.open(in_fn)
active_object = native_file.ActiveObject
active_shape = active_object.Shape.copy()

active_mesh = native_file.addObject("Mesh::Feature", "Mesh")
active_mesh.Mesh = MeshPart.meshFromShape(Shape=active_shape,
                                          # MaxLength=1, # Mefisto
                                          LinearDeflection=int(1.0 * 10),  # Standard
                                          AngularDeflection=math.radians(5),  # Standard
                                          Relative=False
                                          )

Mesh.export([active_mesh, ], out_fn)

App.getDocument(App.ActiveDocument.Label).removeObject(active_mesh.Label)

App.closeDocument(App.ActiveDocument.Label)

sys.exit(0)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39153 (Git)
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Defeaturing 1.2.2
  * fasteners 0.5.31
  * fcgear
  * kicadStepUpMod 11.4.1
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-12-28:
Always wrap your python code in the code tag (left and right angle brackets) to preserve the indentation.

oktabruna, 2024-12-29:
ChatGPT fix my script

closed 2024-12-29 by oktabruna