FreeCAD/FreeCAD #18785: [CORE] Software to pause for seconds when a face is selected

luisDias89, 2024-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem is as follows: whenever I select a face, the software pauses for a few seconds, and depending on the geometry, it can be almost half a minute. Then the following message appears at the end (Image below)
[image]

File: 
[Rat Rig V-Core_fostTesteGit.zip](https://github.com/user-attachments/files/18267928/Rat.Rig.V-Core_fostTesteGit.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Portugal (pt_PT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.34
  * OpenTheme 2024.9.1
  * sheetmetal 0.5.6
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luisDias89, 2024-12-28:
Last week's version did not have this issue

FreeCAD_weekly-builds-39708-conda-Windows-x86_64-py311

luisDias89, 2024-12-30:
@maxwxyz @yorikvanhavre Please check what might be causing this. The software keeps freezing constantly when select someting in helix or some more complex geometry, The builds-39503 version does not have this issue. 
It was added recently

Video with problem:
https://github.com/user-attachments/assets/4cc120e0-1e4d-4a03-b449-a8bc7edcd874

I think it is because of this commit made by @WandererFan
[image]

luisDias89, 2024-12-30:
Build 39503 its OK
https://github.com/user-attachments/assets/3f15a841-1974-421c-a529-f9364816a6d7

Syres916, 2025-01-04:
@WandererFan I'm inclined to agree with @luisDias89 that https://github.com/FreeCAD/FreeCAD/commit/abb50a4daa43abee9e6b263dd82e5e876c978c6a appears to be a little too expensive especially when curved faces are involved. as you can see from the screenshot there are no issues with the geometry of the face and yet there's no response for just over 2 seconds on my self compiled box.

[image]

It is no different in Safe Mode so I'm happy my personal settings are no having an effect here. Would it be possible for you to run it through a debugger please to find the issue? My crude debugging suggests it's doing the `getSubTopoShape` more than once which maybe correct but doesn't 'feel' right when there's only one face.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39757 (Git)
Build type: Release
Branch: main
Hash: a974e413fa39bbfa2572e99bc3af3824ad483263
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.3
  * sheetmetal 0.6.13
  * Behave-Dark-Colors 0.1.1
  * FreeCAD_SketchArch
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.34
  * Curves 0.6.53
  * Assembly4 0.50.15
  * A2plus 0.4.68

```

furgo16, 2025-01-05:
I can also reproduce the part where the `Measurement::radius - Invalid References3D Provided` is shown now, when using the sample file in https://github.com/FreeCAD/FreeCAD/issues/18839.

Simply select one of the cylindrical columns and the message will appear. My model is simple enough that I don't notice a delay, though. 

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Snap 1247
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
```

Syres916, 2025-01-07:
@WandererFan if you want a real slowdown (5 minutes) to test then the model to use is on the [forum](https://forum.freecad.org/viewtopic.php?t=93366#p800212)

clytle374, 2025-01-10:
I'm having the same issue when trying to chamfer or radius in the current git code.  Strangely the others edges all chamfer or radius fine, just this one body fails.  Willing to test things if I can help 
[image]

`OS: Gentoo Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39802 (Git)
Build type: Release
Branch: main
Hash: 558eb19103518090d494fcde0e8deabba830f734
Python 3.12.8, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.31
  * kicadStepUpMod 11.4.1
`

chennes, 2025-01-14:
Should be closed by https://github.com/FreeCAD/FreeCAD/pull/19032

closed 2025-01-14 by chennes