FreeCAD/FreeCAD #18787: Fully constrained geometries displayed and act unconstrained

1984-git, 2024-12-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**How to reproduce:**
1. Create a new parametric body
2. Create a new sketch
3. Draw a circle from the origo with the radius of 10mm
4. (at this point the circle is fully constrained)
5. Draw a line from the origo to the circle (like you draw a radius)
6. Now both the line and the circle shows up as unconstrained
7. Click on the circle without releasing the button and make a dragging move (the circle will not move as its center is constrained to the origo)
8. Release the mouse button
9. Now the circle displays as constrained again.

This already shows one of the problems.

**Another issue:**
1. Close everything
2. Open freecad
3. New prametric body, new sketch
4. Draw two circles from the origo with the diameter of 10mm and 20mm
5. Both circles are constrained
6. Draw two lines from the origo to the outer circle
7. Everything becomes unconstrained
8. Now try to use the trim edge tool on the outer circle. I'd expect this to trim the edge between the intersecting points of the lines.
9. Instead of this it completely removes the whole line
10. Undo the trim
11. Do the dragging move described above on any of the circles
12. Both circles displayed as constrained
13. Repeat the trim edge: **it work just fine now**
14. To demonstrate the next problem reverse the cut trim (so circles became constrained again)
15. Save the body
16. Close FC

**Another issue:**
1. Open the document you just saved
2. Open the sketch
3. Now everything shows up as unconstrained again
4. As everything looks exactly like in (7) of the previous list I'd expect that a trim edge on the outer circle would do the same and remove the whole circle
5. Use the trim edge tool on the outer circle
6. It cuts only the part you really want to cut, however everything remains unconstrained

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-29:
Please create individual issues and check if similar exist. Ideally provide the demo files and record a macro to provide the steps what you are doing. You can edit this issue to reflect one issue.

1984-git, 2024-12-29:
I think these are most likely the different appearance of the same problem, but by all means very strongly related. For me it doesnt make sense to separate them. (note: slightly edited to post not saying "another issue" but "another variation")

I've uploaded a zip file, which partially demonstrates the problem. The file contains a readme file with some description.

I can't provide a full demo as it appears when the geometries are createdbut the steps in the issue are very easy and very quick to follow anways.

[trim-test.zip](https://github.com/user-attachments/files/18269572/trim-test.zip)

1984-git, 2024-12-29:
A video is recorded about creating a fully constrained circle which gets unconstrained by drawing a radius line and then gets fully constrained again by doing the dragging move I explained in the issue description.

https://github.com/user-attachments/assets/f0cab85f-21dc-4f96-9dc6-773489dffb55

GeneGH, 2024-12-30:
I can verify this behavior if I open the file provide by the OP on the forum. However, I cannot verify if I start with a new file I create following the given recipe.

Possibly some difference in settings.

Exactly the same OS and FC version as the OP.

1984-git, 2024-12-30:
Demonstrating the 2nd variation. I do the dragging move around 20s and it takes about 10-12 sec for freecad to figure that the circles are constrained, so not much happens in this range of the video.

https://github.com/user-attachments/assets/4d453b84-0be9-4a11-8ac3-c953fc80bf04

AjinkyaDahale, 2025-03-13:
Could someone confirm whether or not this happens in 1.0?

1984-git, 2025-03-14:
> Could someone confirm whether or not this happens in 1.0?

Its still there in 1.1

luzpaz, 2025-03-15:
@1984-git please add your full [About](jttps://wiki.freecad.org/About) info. TIA!

1984-git, 2025-03-17:
> [@1984-git](https://github.com/1984-git) please add your full About info. TIA!

here we go:

`OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Europe (en_150)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
`

1984-git, 2025-10-31:
The problem is not present anymore in:

`OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/07 22:49:55
Build type: Release
Branch: main
Hash: 68f083a6d169ec523fa9d46b7cc40addb1e4ffbe
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Europe (en_150)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1
`

closed 2025-10-31 by 1984-git