FreeCAD/FreeCAD #18788: large negative values in offset geometry sketcher tool cause CTD

poulet67, 2024-12-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- create a new drawing
- sketch a small square in any plane, then select it
- select "offset geometry" in sketcher workbench
- enter a large negative value in the number box (or at least larger than theoretically possible with the given square)
- crash to desktop

### Full version info

```shell
1.0.0RC4
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-29:
Please test with stable 1.0 and weekly 1.1 https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2025-08-22:
@poulet67 ping