FreeCAD/FreeCAD #18793: TechDraw: ANSIB.svg white background

pmjdebruijn, 2024-12-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is really minor, however...

ANSIB.svg has a white background, while no other template has this... (others are transparent)

And ANSIB_Landscape.svg seems identical to ANSIB.svg except for not having the white background...

(BTW, are SVGs in FreeCAD somehow normalized (for nicer diffs)? for example using scour or something?)

### Full version info

```shell
1.0.0+41~gd9e8fe0d7d
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-12-30:
IIRC, ANSIA and ANSIB predate the rest of the ANSI templates.   I guess they were kept around for backwards compatibility. 

There is no process in place to clean up templates with scour automatically.   They should all probably be resaved as plain svg after running through scour.

WandererFan, 2024-12-30:
There is definitely something funky about ANSIB.svg.  I tried to remove all the white backgrounds - they aren't there in inkscape, but they are there in td.  May need a line by line review of the svg.

Another note, running ANSIB.svg through scour moved the "weight" text.  Another line by line check to see what happened, I guess.