FreeCAD/FreeCAD #18794: [TD]Reading Svg file containing non-ASCII chars fails on mac

WandererFan, 2024-12-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On macOS, inserting an Svg Symbol fails if the Svg contains non-ASCII characters. 

[image]

### Full version info

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39741 (Git) Conda
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Reinforcement
  * Movie 2024.2.15
  * freecad.gears 1.3.0
  * fasteners 0.5.33
  * Rocket 4.0.1
  * Ship 2024.11.26
  * sheetmetal 0.6.12
  * MakerWorkbench 1.0.1
  * Curves 0.6.52
  * Assembly3 0.12.2
```

### Subproject(s) affected?

TechDraw

### Anything else?

Problem reported in forum thread: https://forum.freecad.org/viewtopic.php?t=88142

This is likely due to the following code:
```
        openCommand(QT_TRANSLATE_NOOP("Command", "Create Symbol"));
        doCommand(Doc, "f = open(\"%s\", 'r')", (const char*)filespec.c_str());
        doCommand(Doc, "svg = f.read()");
        doCommand(Doc, "f.close()");
```
Possible solution: use codecs.open with a utf-8 encoding.

Problem: lack of macOS test facilities.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-12-30:
@WandererFan we can use the CI mac-os test, see https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:TD_Test_Non_Ascii_Symbol_dev the correct error is generated so you can test if the `codecs.open` works.

luzpaz, 2024-12-31:
Just noting that the screenshot is showing FreeCAD v39024 and that contradicts the version info 39741

Syres916, 2024-12-31:
@luzpaz while perfectly true, I can say with some certainty that it will have been the case ever since the Conda mac builds went to Python 3.11. I could probably submit a PR tomorrow assuming the tests pass with the new code.

FC-FBXL5, 2024-12-31:
Error message from latest weekly build is still the same:

`pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/kay/Apps/FreeCAD-1-1-0-39748.app/Contents/Resources/lib/python3.11/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'UnicodeDecodeError'>: 'ascii' codec can't decode byte 0xc3 in position 841: ordinal not in range(128)
`

Syres916, 2025-01-01:
@FC-FBXL5 hopefully in the next few days as long as @WandererFan is happy with #18816 then you should have this fixed as the error reported in the CI tests is now gone with the PR.

FC-FBXL5, 2025-01-04:
Nothing changed in weekly build FreeCAD v1.1.0-39750.

WandererFan, 2025-01-04:
The next PR review meeting is 2024-01-07, so this won't be available until then next weekly build after that.

closed 2025-01-06 by yorikvanhavre