FreeCAD/FreeCAD #18796: PartDesign: Hole: Creation of blind threaded hole fails with "removing splitter failed"

7th-fret, 2024-12-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create a simple sketch and pad it, say 20mm.
Then select the top face and use the hole tool to create a blind hole with threads and depth of say 15mm.
It fails with "removing splitter failed"
If the hole length is all the way through the body, it success.

I tried the same procedure with FreeCAD 1.0.0 and it success.

[image]

### Full version info

```shell
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39747 (Git)
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.13.1, Qt 5.15.16, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * CurvedShapes 1.0.13
  * OpticsWorkbench 1.0.26
  * A2plus 0.4.68
  * Curves 0.6.53
  * sheetmetal 0.6.13
  * OpenTheme 2024.5.16
  * freecad.gears 1.3.0
  * fasteners 0.5.34
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-29:
Confirm
Can be workaround if set Refine = False

<details>
  <summary>FreeCAD 1.1.0dev.39741 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

</details>

maxwxyz, 2024-12-29:
@FlachyJoe with all those splitter errors, what about checking the refine property and propose a workaround in the error message itself?

7th-fret, 2024-12-29:
> Confirm Can be workaround if set Refine = False
> 
> FreeCAD 1.1.0dev.39741 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)

Usually i have auto refine enabled in PartDesign settings.
With hole's dialog still open, if i set refine to false in data of hole feature, the threaded hole success.
So, at this moment this workaround seems to work.

[image]

FlachyJoe, 2024-12-29:
@maxwxyz the refine failure could raise a warning instead of an error as I do for helix in #16877. Oddly enough, no one complains. :open_mouth:

closed 2025-03-08 by maxwxyz