FreeCAD/FreeCAD #18798: Problem with fillet operation

ZbiguZbigu, 2024-12-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[Fillet issue.zip](https://github.com/user-attachments/files/18270911/Fillet.issue.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-29:
Constant problems with circles.
To workaround set circle little bit bigger than Pad.

[image]

maxwxyz, 2025-01-02:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/5561

closed 2025-01-02 by maxwxyz