FreeCAD/FreeCAD #18799: Cannot start FreeCAD 1.0.0 AppImage

nvdl, 2024-12-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Downloaded the latest version from https://www.freecad.org/downloads.php.
URL: https://github.com/FreeCAD/FreeCAD/releases/download/1.0.0/FreeCAD_1.0.0-conda-Linux-aarch64-py311.AppImage

But it aborts when trying to run it from the command line.

```
chmod +x FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage
./FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage
```

```
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

AttributeError: module 'shibokensupport.signature.loader' has no attribute 'finish_import'
Fatal Python error: init_phase_2: could not initialize part 2
Python runtime state: initialized

Current thread 0x00007fa821d44840 (most recent call first):
  File "/home/nvd/.local/lib/python3.11/site-packages/shiboken2/__init__.py", line 30 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "/tmp/.mount_FreeCAmiDdMj/usr/lib/python3.11/site-packages/PySide2/__init__.py", line 31 in _setupQtDirectories
  File "/tmp/.mount_FreeCAmiDdMj/usr/lib/python3.11/site-packages/PySide2/__init__.py", line 80 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "/tmp/.mount_FreeCAmiDdMj/usr/Ext/PySide/__init__.py", line 2 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "/tmp/.mount_FreeCAmiDdMj/usr/Mod/AddonManager/AddonManager.py", line 35 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "/tmp/.mount_FreeCAmiDdMj/usr/Mod/AddonManager/InitGui.py", line 6 in <module>
  File "<string>", line 142 in RunInitGuiPy
  File "<string>", line 205 in InitApplications
  File "<string>", line 283 in <module>

Extension modules: FreeCAD, FreeCADGui, shiboken2.shiboken2, PySide2.QtCore (total: 4)
/tmp/.mount_FreeCAmiDdMj/AppRun: line 31: 113684 Aborted                 "${MAIN}" "$@"
```

OS: Debian GNU/Linux 12
Kernel: Linux 6.1.0-28-amd64

### Full version info

```shell
Cannot as it does not reach that point.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-30:
You wrote url for aarch64
command for x86_64
check what you really do?

nvdl, 2024-12-30:
Updated. The kernel version is at the end of the text coming from "uname -a". Or, do you need the whole output of that command?

tarman3, 2024-12-30:
Please try last dev build
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

nvdl, 2024-12-30:
The last dev build has the same issue. It looks like it picks up my local installation of "shiboken2" instead of the packaged one.

Running:

`pip3 uninstall --break-system-packages shiboken2`

has fixed the issue.

```
Found existing installation: shiboken2 5.13.2
Uninstalling shiboken2-5.13.2:
  Would remove:
    /home/<user>/.local/lib/python3.11/site-packages/shiboken2-5.13.2.dist-info/*
    /home/<user>/.local/lib/python3.11/site-packages/shiboken2/*
Proceed (Y/n)? 
  Successfully uninstalled shiboken2-5.13.2
```

oursland, 2024-12-30:
This appears to be a packaging issue.

@luzpaz could you move this issue to the FreeCAD-Bundle repo?