FreeCAD/FreeCAD #18801: Mesh: gmshExe empty by default

pmjdebruijn, 2024-12-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently the gmshExe property is empty by default, this means gmsh meshes will never "just work" without the user pointing FreeCAD explicitly to a binary.

However, for Linux/BSD it can fairly safely be assumed to be in PATH, so just defaulting it to 'gmsh' will typically work fine, and in the rare cases it's not, the user can still point manually to the right binary, so the end result still won't be worse as it is now.

### Full version info

```shell
1.0.0+41~gd9e8fe0d7d
```

### Subproject(s) affected?

Mesh

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-02:
please add full about info.
@FEA-eng FYI

pmjdebruijn, 2025-01-02:
I think this is the relevant code:
  https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Mesh/Gui/RemeshGmsh.ui#L141
  https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Mesh/Gui/RemeshGmsh.cpp#L186

So the issue still applies to main as far as I can tell.

@maxwxyz I'm not sure how my extended version info is relevant here...

luzpaz, 2025-01-03:
@pmjdebruijn from experience it's always worth adding full about info. Especially for posterity. Thanks in advance!

PR #23468 (2025-09-01): Mesh: fix #18801 - simply call `gmsh` if gmsh executable path is empty
fix #18801 by just calling `gmsh` so the user doesn't need to manually provide the executable path, we provide gmsh with most (all?) of our distributions so this gives a better user experience
commit ff91ce14 (2025-09-01): Mesh: fix #18801 - simply call `gmsh` if gmsh executable path is empty
closed 2025-09-01 by chennes