FreeCAD/FreeCAD #18806: Strange behavior Toggle freeze elements of PartDesign body

tarman3, 2024-12-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Click **Toggle freeze** for body object.
The elements inside are also frozen.
It's not clear how to get everything back.

https://github.com/user-attachments/assets/f6258901-8fad-4edc-8d88-e168f66a5969

Another example.
I get some project with CAM.
Used **Toggle freeze** for body object.
Get frozen elements in project which not so easy to find.
Also can`t save project while some elements frozen

https://github.com/user-attachments/assets/20e8d0b2-6e2c-494e-b187-94792e549ed1

First time i found with problem and found all elements inside tree and unfreeze all elements. But now feature **Toggle freeze** disappear for elements inside body.

### Full version info

```shell
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39741 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.12; Silk 0.1.5; Curves 0.6.52
```

### Subproject(s) affected?

None

### Anything else?

[VacuumTable.zip](https://github.com/user-attachments/files/18277705/VacuumTable.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-30:
Probably related to https://github.com/FreeCAD/FreeCAD/pull/12856

tarman3, 2025-01-01:
Found way to unfreeze elements inside Body.
Need to select Body first and after all other elements to which bindings are created.

https://github.com/user-attachments/assets/13fcf372-db2e-4ed3-bf81-ff8a116e012a

maxwxyz, 2025-01-01:
@FlachyJoe FYI

FlachyJoe, 2025-01-04:
Hi!
I am not convinced by the new behavior of Toggle_Freeze. First we can't freeze a feature, change a parent and unfreeze to recompute because the parent is frozen by the toggle. So the frozen state is quite useless.

For this peculiar issue a toggle_freeze on the body should unfreeze all its features.

vletii, 2025-03-22:
Hi, I’m working on a fix for this issue and wanted to clarify the expected behavior. Should the automatic unfreezing of inner elements only apply when unfreezing a "Body" type, or should it apply to any element that contains inner elements?

tarman3, 2025-03-23:
Probably only the chosen element should freeze/unfreeze
It is unclear why something is freeze inside of the chosen Body

vletii, 2025-03-23:
One possible reason for freezing inner elements is to make sure the entire object stays locked, preventing accidental changes. 
Should this behaviour be kept, or should freezing/unfreezing only apply to the selected element without affecting its children?

tarman3, 2025-03-23:
At the current stage, any solution is suitable.
Probably other developers will connected to discussing the issue after the appearance of **Pull Request**

PR #20442 (2025-04-03): GUI: Fix Toggle Freeze Behavior
Previously, the `StdCmdToggleFreeze` command only unfroze objects using `getInListRecursive()`, meaning that unfreezing an element would only affect itself or the elements that contained it.
This caused issues when using Toggle Freeze for a Body object, as it made it difficult to identify all frozen elements in a project, and the presence of frozen elements prevented the project from being saved.

Cause of the Bug
The unfreeze operation only considered outward dependencies, meaning that elements further down the hierarchy (inward dependencies) were not properly  [... cut]
commit 72c1659e (2025-04-03): GUI: Fix #18806 Toggle freeze behavior

Previously, StdCmdToggleFreeze only unfroze child objects using
getInListRecursive(). This update ensures that dependent objects
are properly unfrozen.
Freezing behavior remains unchanged.
closed 2025-04-03 by chennes