FreeCAD/FreeCAD #18808: Revolute joint failure

samriddle-debug, 2024-12-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Simple assembly with several revolute joints breaks when moved.
Attaching assembly file.
Video presenting the issue available here: [reddit post with video](https://www.reddit.com/r/FreeCAD/comments/1hpqw9m/assembly_issue_with_revolute_joint/)
[blokada_demo.zip](https://github.com/user-attachments/files/18278013/blokada_demo.zip)

### Full version info

```shell
Official 1.0 release, as well as latest nightly build rev39109 (2fcc531)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-02:
PLease add full about info

FEA-eng, 2025-09-20:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2025-09-26:
@PaddleStroke is this ticket on your radar ?

PaddleStroke, 2025-11-25:
[image]

This assembly is overconstrained.
@aiksiongkoh look at this assembly with multiple revolute joints. Is there a way the solver could ignore the redundancy in such a case?