FreeCAD/FreeCAD #18811: TechDraw: Balloon Annotation black background in PDF export

pmjdebruijn, 2024-12-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding a balloon annotation in TechDraw everything looks fine in FreeCAD itself, however when exporting to PDF, the background of the balloon itself turns black (instead of transparent).

### Full version info

```shell
FreeCAD 1.0.0+41~gd9e8fe0d7d
Qt 5.15.8
```

### Subproject(s) affected?

TechDraw

### Anything else?

[Unnamed-Page.pdf](https://github.com/user-attachments/files/18282140/Unnamed-Page.pdf)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pmjdebruijn, 2024-12-31:
This only happens with File -> Save As
This does not happen with Right click -> Export PDF

pmjdebruijn, 2024-12-31:
File -> Save As produces a PDF/A file
whereas Right click -> Export PDF produces a PDF-1.4 file

So this is likely the transparency removal for PDF/A compliance that's causing the issue.

WandererFan, 2025-01-01:
This is a duplicate of #18443.  PR #18507 (merged  2024-12-16) fixes it.

pmjdebruijn, 2025-01-01:
Excellent! Sorry for the dup... 

Aren't these candidates for cherry picking to releases/FreeCAD-1-0 ?

closed 2025-01-02 by maxwxyz

gazzatav, 2025-05-22:
I just noticed something like this in 1.0.1.  Not black but grey shading has appeared in the backgrounds of balloon annotations when printing a TechDraw page to pdf.  If I export the page as pdf it's even worse with faces and holes being shaded.

OS: Ubuntu 22.04.5 LTS (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.50.18
  * fasteners 0.5.38
  * ExplodedAssembly
  * Silk
  * Curves 0.6.62
  * sheetmetal 0.7.22

[image]

WandererFan, 2025-05-28:
see https://github.com/FreeCAD/FreeCAD/pull/18507#issuecomment-2910305404