FreeCAD/FreeCAD #18814: Sketcher: Ghost Control Points in Spline

dragazo, 2025-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've been playing around with more spline things as possible workarounds to #18250. I've found that if I create a spline by control points and then delete the knots from the elements list, I get the behavior I want where I can move individual control points without affecting the others. Apparently the presence of the knot vertices is what causes things to move for some reason.

However, there is no way to add control points directly - only a way to add knots. So I tried adding a knot and then re-deleting the knots (as they all get added back). However, this somehow results in a seemingly invalid spline:

[image]

From the curvature comb above, there seem to be two ghost knots between the two top-left control points (giving rise to the short almost-solid green part of the curvature comb), which seems like it should not be possible unless there are also some other ghost control points?

Project file (just renamed to `.zip`): [weird-spline.zip](https://github.com/user-attachments/files/18285838/weird-spline.zip)

Since I'm not having luck using splines this way, is there just an option somewhere where I can have a polyline and just apply a modifier to treat it as the control polygon of a spline (while still being able to edit the polyline as a polyline)? Cause then I can just use the split edge tool to add a new control point to the polygon, which is exactly what I want. So far, I've just found freecad splines kind of hard to work with directly, but I really just want to smooth out a control polygon.

### Full version info

```shell
OS: Linux Mint 20 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-02:
@AjinkyaDahale could you take a look at this?

dragazo, 2025-01-03:
Sure, when I add the spline by control points, it also adds knots (which makes sense, cause they're a mathematical necessity for this type of spline). Those show up in the element viewer as vertices, as opposed to control points which are circles. So I just select all of the vertices and delete them. The spline is still fine and can be manipulated via the control points. The reason I do this is because unless they are deleted, moving a control point moves other control points, which to me defeats the purpose of "control" and prevents me from doing local edits of the shape. The same issue is present for moving knots). Also, the knots don't seem necessary since they seem to be controlled by the control points (which makes sense). The shape doesn't even change in the slightest when they are deleted.

And to be clear, I believe the knots are still "there" after I "delete" them, cause they're a mathematical necessity for this type of spline. But their position is determined by the control points, and the presence of the vertex marking their location seems to cause the non-local edits issue I mentioned. I'm really just deleting the vertex marking the location of the knot.

Shouldn't it just be possible to insert a new control point by splitting an edge in the control polygon? That shouldn't have a dramatic effect on the spline, and its effect should not extend beyond its knot-delimited segment.

As a side question, why does the sketcher not have a type of B-spline that is only defined by knots (and no control points, at least no user-controllable ones)? If I'm understanding it correctly, that's exactly what happens when you do a loft. It defines a bunch of splines using vertices of the loft profiles as knots, and then just does it - no need to edit control points or anything - it just figures it out from the knots. That's the type of spline I'm used to editing. Or close, anyway. I'm used to the onshape type where there are N knots and only a single control point on each the terminal end for setting the terminal slopes. But having the loft type of splines in the sketcher would be way closer to what I'm used to - I don't know how involved that would be to add to the sketcher, but the code is at least somewhere already in the codebase since it's used for lofts.

AjinkyaDahale, 2025-01-04:
> [The knot points] show up in the element viewer as vertices, as opposed to control points which are circles. So I just select all of the vertices and delete them.

That only deletes the physical representation of the knot, not remove it from the "knot vector" [1] of the B-spline. Think of the focal points or axes of an ellipse for example. You can delete them, but the fact that they are part of the ellipse does not change. Coming back to the knot, you will see that the curvature comb does not change if you try to delete the knot point this way.

You can indeed delete the knot from the knot vector by using the "[Decrease knot multiplicity](https://wiki.freecad.org/Sketcher_BSplineDecreaseKnotMultiplicity)" command. However, you will see that the B-spline changes shape much more drastically than before.

> ...The reason I do this is because unless they are deleted, moving a control point moves other control points, which to me defeats the purpose of "control" and prevents me from doing local edits of the shape. The same issue is present for moving knots).

Thanks for this insight. 

> Also, the knots don't seem necessary since they seem to be controlled by the control points (which makes sense). The shape doesn't even change in the slightest when they are deleted.

They are perhaps not "necessary" in the sense that they can be completely determined by the "control points" as well as the "knot vector". However, there is presently no physical representation of the knot vector. You can increase or decrease the knot multiplicity 

> And to be clear, I believe the knots are still "there" after I "delete" them, cause they're a mathematical necessity for this type of spline. But their position is determined by the control points, [...]

From this paragraph, it appears you get the theory of B-splines, but are probably conflating some concepts. I'm sorry but the theory is kind of out of scope for this issue.

> Shouldn't it just be possible to insert a new control point by splitting an edge in the control polygon? That shouldn't have a dramatic effect on the spline, and its effect should not extend beyond its knot-delimited segment.

You may try to convince yourself about what happens when you add a control point while keeping all others the same. For this, simply create a B-spline with coincidence constraints on all the new control points except for the one in the middle. Here is an attached file for reference.

[bspline-with-one-added-knot.zip](https://github.com/user-attachments/files/18306698/bspline-with-one-added-knot.zip)

[image]

Adding a control point forces the addition of a knot parameter somewhere in the knot vector. The above makes it such that all knot intervals are still equal, but other methods may be possible, giving different degrees of "dramatic" changes. That is however, not supported with an easy GUI command at the moment.
 

[1] Mind you, it is not a vector in the sense of vector algebra, but more like a non-decreasing sequence of real numbers.

AjinkyaDahale, 2025-01-04:
> As a side question, why does the sketcher not have a type of B-spline that is only defined by knots (and no control points, at least no user-controllable ones)?

That is a spline, but not in a B-spline representation. As such, what we (perhaps colloquially) call "B-spline" is only one representation of the intended curve. There are other representations which may be useful in other contexts.

> If I'm understanding it correctly, that's exactly what happens when you do a loft. It defines a bunch of splines using vertices of the loft profiles as knots, and then just does it - no need to edit control points or anything - it just figures it out from the knots.

I'm not aware of what exactly goes along there, but there it may be more useful to see it as a spline in the shape space rather than the 3-dimensional Euclidean space. This may be yet another "out-of-scope" thing.

> I'm used to the onshape type where there are N knots and only a single control point on each terminal end for setting the terminal slopes. 

I'm not aware of the "onshape type of knots". Perhaps they are in a [centripetal Catmull-Rom](https://en.wikipedia.org/wiki/Centripetal_Catmull%E2%80%93Rom_spline#) form. I do wish to add support for these (and other) types of splines, but it will take time.

> But having the loft type of splines in the sketcher would be way closer to what I'm used to - I don't know how involved that would be to add to the sketcher, but the code is at least somewhere already in the codebase since it's used for lofts.

Not sure if it is, but it may be similar. There is also something similar going on when one uses "create B-spline by knots". In any case, the changes will perhaps need to be made in OCCT itself first before anything can be done in FreeCAD.

AjinkyaDahale, 2025-09-29:
Feature request from what I understand. But what feature exactly I'm not sure.