FreeCAD/FreeCAD #18817: Opening file 0.21.2 in FreeCAD 1.0 loses part orientation

PabloBeranger, 2025-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Document in FreeCAD 0.21.2
[image]

When opening the document in FreeCAD 1.0 the part is no longer in its reference axis.
[image]

When I do the recomputation for migration FreeCAD 0.21.2 to 1.0,
The recomputation rotates the part reference plane and axis and the part is in a completely wrong coordinate system.

[image]

Attached is my original file of FreeCAD 0.21.2
[ACA_Crosstie_Long__NSS.zip](https://github.com/user-attachments/files/18286102/ACA_Crosstie_Long__NSS.zip)

I find that these migration problems are not admissible

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools (Disabled)
  * A2plus 0.4.68
  * addFC 1.0.2
  * AirPlaneDesign 0.4.1 (Disabled)
  * Alternate_OpenSCAD 1.0.0
  * AnimationFreeCAD 1.0.0
  * ArchTextures
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * assembly_handbook
  * BCFPlugin 1.0.0
  * BillOfMaterials 1.0.2
  * BIM021 2021.12.0
  * BIM0212 2021.12.0 (Disabled)
  * BIMBots
  * btl 0.9.9 (Disabled)
  * CadbaseLibrary 1.0.3 (Disabled)
  * cadquery_module 2.0.0 (Disabled)
  * Cfd (Disabled)
  * CfdOF 1.29.4
  * CommandPanel (Disabled)
  * CurvedShapes 1.0.13 (Disabled)
  * Curves 0.6.53
  * Defeaturing 1.2.3
  * Design456 0.0.1 (Disabled)
  * DesignSPHysics 0.7.1(26-11-2024)
  * dodo 1.0.1 (Disabled)
  * dxf-library (Disabled)
  * DynamicData 2.70.0
  * EM 2.1.1 (Disabled)
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.5.34
  * Fcmcua 1.0.0
  * FeedsAndSpeeds 0.6.0 (Disabled)
  * FEMbyGEN 2.3.0
  * FEM_FrontISTR 0.1.0
  * frame 0.1.1
  * freecad.gears 1.3.0
  * GDML 2.0.1Beta (Disabled)
  * Glass (Disabled)
  * Help 1.0.3
  * ImportNURBS 1.1.0 (Disabled)
  * InventorLoader 1.5.1 (Disabled)
  * kerkythea (Disabled)
  * kicadStepUpMod 11.4.4 (Disabled)
  * lattice2 1.0.0 (Disabled)
  * Launcher (Disabled)
  * LCInterlocking
  * Lithophane (Disabled)
  * Manipulator 1.5.7
  * MeshRemodel 1.10.35
  * MnesarcoUtils 0.2.14
  * Movie 2024.2.15 (Disabled)
  * NavigationIndicator (Disabled)
  * Nodes 0.1.36
  * Nodes.backup1723381727.939789 0.1.36
  * nurbs (Disabled)
  * OpticsWorkbench 1.0.26 (Disabled)
  * OSAFE 2022.5.29 (Disabled)
  * ose-piping (Disabled)
  * ose-workbench-core
  * pcb 6.2023.1 (Disabled)
  * persistenttoolbars (Disabled)
  * PieMenu 1.11.2
  * POV-Ray-Rendering
  * PVPlant
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Reinforcement (Disabled)
  * Render 2024.12.15
  * Reporting
  * retr3d (Disabled)
  * SearchBar 1.2.1
  * sheetmetal 0.6.13
  * Ship 2024.11.26
  * Silk 0.1.5 (Disabled)
  * slic3r-tools (Disabled)
  * SteelColumn (Disabled)
  * symbols_library
  * TitleBlock 0.5.2.2
  * toSketch 1.0.1
  * Trails 2022.1.0 (Disabled)
  * WebTools 1.0.0
  * woodworking 0.21.2.33771.0.23.1
  * workfeature
  * workfeature-macro
  * yaml-workspace 0.1.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-04:
@wwmayer will your PR fix the orientation issue of this demo file?

PabloBeranger, 2025-01-04:
What is PR ??

FlachyJoe, 2025-01-04:
> What is PR ??

Pull Request

FEA-eng, 2025-09-20:
I get recompute errors when opening this file in 1.1:

```
12:37:54  <Exception> File 'C:/Users/PC/Desktop/ACA_Crosstie_Long__NSS/ACA_SpreadSheet_AC_Unit_Cover.FCStd' does not exist!
12:37:54  Exception opening file: C:/Users/PC/Desktop/ACA_Crosstie_Long__NSS/ACA_SpreadSheet_AC_Unit_Cover.FCStd [File 'C:/Users/PC/Desktop/ACA_Crosstie_Long__NSS/ACA_SpreadSheet_AC_Unit_Cover.FCStd' does not exist!]
12:37:54  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
12:37:54  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
12:37:54  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
12:37:54  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
12:37:55  <Exception> ObjectIdentifier.cpp(1456): Document 'ACA_SpreadSheet_AC_Unit_Cover' not found in 'ACA_SpreadSheet_AC_Unit_Cover#<<params>>.Mount_S'
in property binding 'ACA_Crosstie_Long__NSS#Sketch022.Constraints'
12:37:55  Sketch022: Document 'ACA_SpreadSheet_AC_Unit_Cover' not found in 'ACA_SpreadSheet_AC_Unit_Cover#<<params>>.Mount_S'
in property binding 'ACA_Crosstie_Long__NSS#Sketch022.Constraints'
```

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

marbocub, 2025-11-17:
Summary: This is not a regression, but a corrupted file reference.
Confirmed on 0.21.2, 1.0.2, 1.1.0dev-weekly-2025.11.12.

---

I investigated this issue and I believe it is caused by data corruption rather than a regression. 
After fixing the data, by using 0.21.2, I was able to open and recompute the file in both 1.0.2 and 1.1.0dev without any problems. 
The orientation remained unchanged, consistent with 0.21.2, and no rotation occurred.

On 0.21.2, the Report View shows three types of messages. I think the first one is the actual problem.

1:
```
<Exception> File 'C:/Users/me/Desktop/freecad-issue-confirmation/18817/ACA_SpreadSheet_AC_Unit_Cover.FCStd' does not exist!
Exception opening file: C:/Users/me/Desktop/freecad-issue-confirmation/18817/ACA_SpreadSheet_AC_Unit_Cover.FCStd [File 'C:/Users/me/Desktop/freecad-issue-confirmation/18817/ACA_SpreadSheet_AC_Unit_Cover.FCStd' does not exist!]
```

Six objects reference ACA_SpreadSheet_AC_Unit_Cover via expressions, but the file is missing, so the references are broken:

* Pad009 (Label: main)
* Pad010 (Label: 1st Tenon001)
* Pocket014
* Sketch022
* Sketch023
* Sketch024

After clearing the expressions and replacing them with values, then saving again, the file can be handled correctly in 1.0.2 and 1.1.0dev. The orientation is the same as in 0.21.2 and does not rotate.

2:
```
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
```

This message appears because the following object references an Edge. I think this is normal information. If LCS_Central_Avant is deleted, the message disappears.

* LCS_Central_Avant

3:
```
PartDesign::CoordinateSystem: Link(s) to object(s) 'Pocket014' go out of the allowed scope 'LCS_1'. Instead, the linked object(s) reside within 'Part Part'.
PartDesign::CoordinateSystem: Link(s) to object(s) 'Pocket014' go out of the allowed scope 'LCS_2'. Instead, the linked object(s) reside within 'Part Part'.
PartDesign::CoordinateSystem: Link(s) to object(s) 'Pocket014' go out of the allowed scope 'LCS_3'. Instead, the linked object(s) reside within 'Part Part'.
PartDesign::CoordinateSystem: Link(s) to object(s) 'Pocket014' go out of the allowed scope 'LCS_4'. Instead, the linked object(s) reside within 'Part Part'.
```

This seems to occur because the following four objects do not belong to Body006 (Label: Crosstie Long):

* LCS_Gauche
* LCS_Droite
* LCS_Central
* LCS_Central_Avant

@PabloBeranger Could you please re-check item 1 above?

@maxwxyz @FEA-eng FYI.

Here is the fixed file for reference (corresponding to item 1 above):
[ACA_Crosstie_Long__NSS-fixed.zip](https://github.com/user-attachments/files/23573429/ACA_Crosstie_Long__NSS-fixed.zip)

marbocub, 2025-11-17:
As an alternative, by placing the following file `ACA_SpreadSheet_AC_Unit_Cover.FCStd` in the same folder where the original data is stored, the original data can also be opened correctly on 0.21.2, 1.0.0, 1.0.2, and 1.1.0dev-weekly-2025.11.12. 

This result further confirms that the issue is neither a bug nor a regression, but rather a problem with the data itself.

File: [ACA_SpreadSheet_AC_Unit_Cover.zip](https://github.com/user-attachments/files/23573935/ACA_SpreadSheet_AC_Unit_Cover.zip)