FreeCAD/FreeCAD #18818: Weekly Build für den Mac Mini M1 defekt

Cus1057, 2025-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Die Weekly Build für den Mac Mini M1 ("ARM" Versionen) funktionieren schon seit längerem nicht mehr.
Es kommt immer die Meldung: "Die Datei ist beschädigt und kann nicht geöffnet werden"
Mittlerweile habe ich es mit 5 verschiedenen probiert und keine funktioniert.
Die Intel Versionen gehen.

### Full version info

```shell
FreeCAD_weekly-builds-39748-conda-macOS-arm64-py311.dmg
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2025-01-01:
https://github.com/FreeCAD/FreeCAD-Bundle/issues/360#issuecomment-2561365902

maxwxyz, 2025-01-02:
see linked issue

closed 2025-01-02 by maxwxyz