FreeCAD/FreeCAD #18819: Inconsistent mode of selection

karamellpelle, 2025-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the _3D view_, select is exclusive so you need to hold down <kbd>Ctrl</kbd> in order to click and select multiple objects. But in _Sketcher_ extended selection is default, clicking a new element will just add to the current selection. This is annoying and creates a bad user experience. Consistent behavior makes easy UI. 

I suggest that _both_ 3D view and sketch use <kbd>Ctrl</kbd>-<kbd>left click</kbd>[^1]. I often click the wrong element (like an edge instead of vertex), so the exclusive select mode is the better choice; the current behaviour is frustrating.

There is also a solution that can benefit both: Use a "Selection mode" radio button as done in Blender with 5 different modes[^2]: New (exclusive), Extend, Subtract, Invert, Intersect, see this picture:

[image]

[^1]: Or maybe <kbd>Shift</kbd>-<kbd>left click</kbd>instead which other 3D-softwares do]
[^2]: This does not seem to work by single click only  if selecting a region, but it's still possible to implement the click behaviour.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * IconThemes
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oficinerobotica, 2025-01-02:
+100 for a "selection mode" option icon in the toolbar for a universal solution. 

In my case I much prefer the sketcher way of selecting. When selecting using CTRL , it is enough for a wrong click in order to lose the selection of say 10 edges to be fileted and one needs to start all over again. It is really annoying and breaks the flow. A selection mode would solve the problem for all use cases and personal user preferences.

maxwxyz, 2025-01-02:
@FreeCAD/design-working-group FYI

karamellpelle, 2025-01-02:
> In my case I much prefer the sketcher way of selecting. When selecting using CTRL , it is enough for a wrong click in order to lose the selection of say 10 edges to be fileted and one needs to start [...]

But <kbd>Ctrl</kbd>-<kbd>click</kbd> that wrong object again should deselect it and let you continue your selection, or it does not do so in FreeCAD? That's usually how selections work, for example in a file explorer.

I'm also annoyed when loosing my selections, for example in Inkscape. In Blender the action of selecting is recorded in the undo stack, so <kbd>Ctrl</kbd>-<kbd>Z</kbd> lets you continue if you lost your selection. Record our selections could actually be a new Github issue! FreeCAD has some setting "_Record selection in tree view in order to go back/forward using navigation button_", but I don't think that's the same.

[image]