FreeCAD/FreeCAD #18820: Show tooltips even when window is not in focus

dzid26, 2025-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While reading forum how to use FreeCAD ;) I often try to hover on FreeCAD icons to read their labels. However this doesn't work because tooltips do not show up when FreeCAD loses focus. 

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * 3D_Printing_Tools
  * CubeMenu
  * Defeaturing 1.2.2
  * fasteners 0.5.33
  * kicadStepUpMod 11.4.4
  * Launcher
  * ModernUI (Disabled)
  * Ondsel-defaults.cfg
  * OpenTheme 2024.9.1
  * PieMenu 1.10.1
  * POV-Ray-Rendering
  * ToolbarStyle
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

philjynx, 2025-01-03:
I foresee a minor problem if this were enabled. If FreeCAD were behind amother app, would the tooltips show if mouse pointer was over them?

dzid26, 2025-01-03:
Of course that is not the intention. It should react only when hovering over visible part of the window. Pretty standard behavior.

luzpaz, 2025-01-03:
You can do this with other applications that aren't in focus ?

dzid26, 2025-01-03:
Of course. 
Chrome, OrcaSlicer, MS Paint, VS Code, etc

Example from OrcaSlicer while the focus was in Chrome:
[image]

I guess the issue is related to the fact that Freecad uses some custom looking tooltips.

xtemp09, 2025-01-06:
This is possible, but, I ought to say, that this is default behavior for _Qt_ applications.[^1]

> By default, all tool windows are hidden when the application is inactive.

[^1]: https://doc.qt.io/qt-6/qt.html

xtemp09, 2025-01-06:
> mouse scrolling are always active

In _Windows_, scrolling of inactive windows is disabled by default to avoid confusion with active windows. Though, something may have been changed recently.[^1]

[^1]: https://www.elevenforum.com/t/turn-on-or-off-scroll-inactive-windows-in-windows-11.6102/

Syres916, 2025-01-11:
@dzid26 it can be done by https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Tooltips_Always_Show_dev leaving the existing behaviour as is but allowing users to have your requirement, obviously there would be a proper checkbox in Preferences for this but I'll leave that for UI pros.

dzid26, 2025-01-11:
@Syres916  That's great. 
But why litter the Preferences with a checkbox for legacy behavior from 20y ago?

There are billions of peoples using Chrome where the browser hints and even whole websites already react to the hovering. This is what everyone expects.

Syres916, 2025-01-11:
@dzid26 you're asking the wrong person, I've given a technical solution, if you want to raise a Pull Request using it as default and have the political discussion with the decision makers then by all means give it a try.

closed 2025-01-27 by yorikvanhavre