FreeCAD/FreeCAD #18828: Tech Draw: spreadsheet view renders improperly when there are merged cells

MiqCG, 2025-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When inserting a Spreadsheet into a Tech Draw page as described in the [wiki](https://wiki.freecad.org/TechDraw_SpreadsheetView), the spreadsheet is not correctly shown if there are merged cells in it. See attached images.

Tested with version 1.0.0 (Windows) and 1.1.0 rev 39708 (Linux). Issue discovered by @SanaBrian.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

TechDraw

### Anything else?

Spreadsheet element:
[image]

Spreadsheet view in Tech draw:
[image]

[Techdraw-Sreadsheet.FCStd.zip](https://github.com/user-attachments/files/18292447/Techdraw-Sreadsheet.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amreo, 2025-01-03:
I have reproduced the issue, and freecad shows the same artifact.

```
OS: KDE Flatpak runtime (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.25
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.9
  * boltsfc 2022.11.5
  * ThreadProfile 1.92.0
  * sheetmetal 0.4.23
  * lattice2 1.0.0
  * Curves 0.6.42
  * Assembly4 0.50.6
  * Assembly3 0.12.2
  * A2plus 0.4.65
```

Steps to reproduce:
1) create a new document
2) create a spreadsheet
3) copy & paste this text to A1
	'lenZ	'lenX	'lenY	'scale
'A	9	8	4	2
'B	9	5		3
'C	1		111	1
4) join B4-C4
5) join C3-D3
6) go to techdraw WB
7) create a new default page
8) add view from spreadsheet
9) change Cell End property of the sheet "object" to E4

[image]
[image]

maxwxyz, 2025-01-03:
@WandererFan FYI

SanaBrian, 2025-01-03:
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

Taken a first example, when I add a extra row changing Cell End property of the sheet "object" to E5, cells combined perfectly, but the inserted table in TechDraw´s sheet show a void row.
[image]