FreeCAD/FreeCAD #18830: BIM Custom Door not working

nkrill46, 2025-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Unable to create simple rectangular custom door in FC 1.0 (I actually wanted to create a door with a rounded top)
ref: https://www.bing.com/videos/riverview/relatedvideo?q=freecad+custom+door&mid=40EF1683DBDFA1254C9F40EF1683DBDFA1254C9F&FORM=VIRE

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/Fusion
Installed mods: 
  * Assembly4 0.50.15
  * Curves 0.6.51
  * fasteners 0.5.31
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-03:
Please add an example file and steps how to reproduce

nkrill46, 2025-01-19:
I haven't found a way to create a door with a round top as found in Spanish architecture, Even the prodigious Freecad parts library doesn't seem to offer a solution??

Roy-043, 2025-01-20:
The problem here is that the bottom edge of the wall is used as an external reference in the base sketch of the door. When a hole for the door is created in the wall, its shape changes, and also the indices of its edges (BIM in not yet 'TNP safe'). What was edge 9 is now edge 10 and edge 18. If you remove the external geometry from the sketch, or reference the edge from the base sketch of the wall, the problem should be solved.

BTW: the example file was created in version 1.1, which is a bit confusing.

closed 2025-01-20 by maxwxyz