FreeCAD/FreeCAD #18833: Freecad weekly builds crashes on MAC OSX

Tungee, 2025-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The weekly builds of freecad does not start on my M1 Mac Sonoma since weeks.
I opened the dmg and copied the Freecad app on my desktop. I rightclicked the app and and click on Open, after that i get a crash message, that freecad is damaged. Here is the snapshot of the message. 
The official 1.0 Freecad mac runs fine nevertheless.

[image]

### Full version info

```shell
Freecad 1.1 39748
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oursland, 2025-01-02:
The weekly builds are unsigned.  Recent versions of macOS have made it such that 'quarantined' files cannot be run, whereas previous versions would provide a warning.  You need to remove the `quarantined` attribute:

`xattr -cr /Applications/FreeCAD.app`

We are investigating ways to automatically sign the weekly builds to remove this annoyance.

Tungee, 2025-01-02:
Thank you very much!

closed 2025-01-02 by Tungee