FreeCAD/FreeCAD #18834: No Text Tool in Sketcher

Will-Stock, 2025-01-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A problem that I have found is a distinct lack of an easy-to-use text tool in the Sketcher Workbench. Being able to add text with the click of a button would be wonderful

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods:
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * pcb 6.2023.1
  * Render 2024.12.15
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-03:
Getting a huge number of lines with different constraints in Sketch is not a good idea

maxwxyz, 2025-01-03:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12944

closed 2025-01-03 by maxwxyz