FreeCAD/FreeCAD #18835: Mousewheel unreactive while moving on Linux Mint 22

Maxhud1, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The mousewheel is unreactive while moving the mouse; zooming in or out while moving will delay the input. (on any 3d navigation that uses mousewheel to zoom)

If the movement stops while still scrolling the mousewheel, the view will update as if all the input loaded at once.
If the scrolling ends before the movement, the inputs seem to store and will be released on the next mousewheel scrolling.

### Full version info

```shell
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?t=93435
https://www.reddit.com/r/FreeCAD/comments/1hs1hxl/mousewheel_unreactive_while_moving/

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-03:
Can you reproduce on the weekly AppImage ?  https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2025-01-03:
cc @Rexbas when you have a moment and if you could be so kind to weigh-in ?

Maxhud1, 2025-01-03:
I've run the AppImage, the issue is almost gone but I can definitely still reproduce the issue. 
Mousewheel is a lot more reactive even when I'm trying hard to move the cursor but there's is still some sticking from time to time but it is a lot more usable as it is.
It has become a minor annoyance instead of being very frustrating.

Maxhud1, 2025-01-04:
I've found that the mousewheel also becomes unreactive when navigating the Parameter Editor, same behavior; input delayed on cursor movement.

Rahix, 2025-01-04:
I think I have the same problem: The first (few) mouse wheel events seem to be ignored, only after scrolling for a while does the action actually pass through.  This happens when zooming in the 3D view and also when scrolling in an input field to change a numeric value.

### Version Info
```text
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

Rahix, 2025-01-04:
Retried with the weekly build linked above, using the AppImage.  This build does **not** have the problem, it seems.  So either this is a Flatpak-specific problem or it was already fixed?  Trying to find out more...

### Version Info
```text
OS: Arch Linux (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

Rahix, 2025-01-04:
Problem is also not reproducible with the [1.0.0](https://github.com/FreeCAD/FreeCAD/releases/tag/1.0.0) AppImage.  So this seems to be a Flatpak-specific issue.

Rexbas, 2025-01-04:
I can confirm this problem with my own build with Qt 6, but I also have some problems with weekly build 39750 AppImage which is Qt 5.

```
OS: Fedora Linux 41 (KDE Plasma) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39757 +1 (Git)
Build type: Release
Branch: fork/Flast/performance/mouse/hover
Hash: 49969d22863edd3575fbe9ce2b432c69c9b5af35
Python 3.12.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
```

Rexbas, 2025-01-04:
Using `setAttribute(Qt::AA_CompressHighFrequencyEvents, false);` fixes this at the cost of performance, especially in big files the performance goes from ok to absolutely unusable so this attribute is not desired. However it does appear to be a X11 specific problem. I have a build with wayland support where this issue does not occur. So we have to wait until there is a coin version with wayland support and then this issue will be resolved. Problem is that the wayland work in coin (https://github.com/coin3d/coin/pull/531) is a bit slow/stuck.

Maxhud1, 2025-01-05:
Wayland doesn't work on my PC and it is way over my ability to figure out why; where do I access those `setAttribute(Qt::AA_CompressHighFrequencyEvents, false);` settings?

Rexbas, 2025-01-05:
We don't have a setting to set this attribute. `setAttribute(Qt::AA_CompressHighFrequencyEvents, false);` is a line of code.

Maxhud1, 2025-01-05:
I was reading about Wayland and I won't pretend I understand completely after a 30m read but while looking into System Monitor and Freecad Memory Maps I saw xorg (wich I believe refers to x11) was deleted.  Am I onto something?
[image]

Rahix, 2025-01-06:
@Rexbas do you think this is a problem with Qt6 X11 support then?  So something that should be reproducible in any Qt6 program using that attribute?

Rahix, 2025-01-06:
Digging deeper, I cannot reproduce this in a naive simple Qt6 application so far.  Also, the documentation of `AA_CompressHighFrequencyEvents` does not mention the mouse wheel at all, only mouse movement events and the like are affected, apparently.

Rexbas, 2025-01-06:
> I saw xorg (wich I believe refers to x11) was deleted

That could well be possible as FreeCAD uses xorg/X11 even if you are running a Wayland desktop session. If you have a Wayland desktop session then this problem still occurs because FreeCAD is running through XWayland, so still X11.

> do you think this is a problem with Qt6 X11 support then? So something that should be reproducible in any Qt6 program using that attribute?

I don't know, maybe the problem only occurs under specific circumstances. I do know that right now we are not always receiving the event from Qt.

Maxhud1, 2025-01-19:
I tested IMWheel and it doesn't seem to help on my side. Thanks for the advice nonetheless!

coreyoconnor, 2025-02-20:
I was encountering this issue using the flatpak distribution on NixOS. Running freecad in wayland worked for me. I used:

```bash
flatpak run --env=QT_QPA_PLATFORM=wayland --socket=fallback-x11 --socket=wayland org.freecad.FreeCAD
```

Maxhud1, 2025-02-26:
After many tries, I couldn't run desktop in Wayland so I removed my 3090 for a 5700xt and reinstalled Linux Mint.
Mousewheels now work as intended in Freecad using Wayland.

Rexbas, 2025-09-02:
Maybe @oursland knows about updating coin. With coin 4.0.4+ it should be possible to have native Wayland support but it may not work with proprietary NVIDIA drivers.

FEA-eng, 2026-03-12:
> So we have to wait until there is a coin version with wayland support and then this issue will be resolved.

So the issue should still be relevant

> OS: Windows 10 build 19045
> Architecture: x86_64
> Version: 1.2.0dev.20260303 (Git shallow)
> Build date: 2026/03/03 22:57:35
> Build type: Release
> Branch: grafted,grafted
> Hash: 637463ac354f093e14dacfc5059eb70c2b1216fd
> Python 3.11.14, Qt 6.8.3, **Coin 4.0.3**, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
> shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
> Locale: Polish/Poland (pl_PL)
> Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
> Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
> Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1

luzpaz, 2026-04-07:
@oursland any progress ?

brandonbeaumia, 2026-08-18:
For 20260812, Coin is showing 4.0.10 in my About > Libraries but FreeCAD is still running via XWayland. The jerky scrolling is still a problem. Forcing Wayland on the 1.1.3 Flatpak fixes it but I don't know how/if you can force it for the AppImage.

@oursland Can the Wayland switch be flipped on the AppImage if Coin was able to be updated?

Edit: #25448 implies it's fixed but I'm not getting Wayland....

AppImage:
```
OS: Bazzite (KDE//usr/share/wayland-sessions/plasma.desktop/wayland)
Architecture: x86_64
Version: 26.3.0dev.20260812 (Git shallow) AppImage
Build date: 2026/08/12 05:41:29
Build type: Release
Branch: (HEAD detached at dc1ef6e88)
Hash: dc1ef6e880d6f5cf5d5a6eb3749373141eb0deb8
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/109.605/1
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.1.6/AMD/AMD Radeon Graphics (radeonsi, renoir, ACO, DRM 3.64, 7.1.8-ogc1.1.fc44.x86_64)
```

luzpaz, 2026-08-30:
bumping for visibility