FreeCAD/FreeCAD #18836: Part WB 2D Offset - Fill Offset bug using B-spline

Nigelkdx200, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a sketch create a simple B-Spline. Then in the Part WB perform a positive 2D Offset of 3mm. The offset creates fine and then tick Fill Offset. This then fills as expected. (See Image 1). However if you make it a negative 2D Offset -3mm while the Offset creates correctly (See Image 2) when you tick Fill Offset it fills the entire shape rather than just the offset. (See Image 3). Adjusting the amount of offset makes no difference. It works fine for any positive offset and fails for any negative offset.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/New Zealand (en_NZ)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

amreo, 2025-01-03:
I tried to reproduce it and freecad behaviour seems to be opposite than yours...
[image]
[image]

```
OS: KDE Flatpak runtime (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.25
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.9
  * boltsfc 2022.11.5
  * ThreadProfile 1.92.0
  * sheetmetal 0.4.23
  * lattice2 1.0.0
  * Curves 0.6.42
  * Assembly4 0.50.6
  * Assembly3 0.12.2
  * A2plus 0.4.65
```

Nigelkdx200, 2025-01-05:
@amreo Your screenshots do appear to be the opposite of what I had however it still confirms that there's an issue. Whether a positive or negative number is used the behavior should be consistent both ways and the fill offset should always be between the shape and offset line and not fill the entire shape.

tarman3, 2025-01-05:
In most case FreeCAD can not create correct offset for bspline.
Try to use [InkScape](https://inkscape.org)

maxwxyz, 2025-01-06:
Related: https://github.com/FreeCAD/FreeCAD/issues/15151