FreeCAD/FreeCAD #18837: Cannot select sketch vertices, eg during datum plane creation

gryphonB-2, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a weird one.
I am trying to select a vertex on a sketch from the main window (ie not in Sketcher), in order to create a datum plane based on it. Which I literally did yesterday. 
In order to do this more easily I increased the size of the vertices in the sketch's Properties, from 4 to 8.
After which it is impossible (as opposed to difficult) to select a vertex. 
I can select any of it's wires, easily.
Changing the vertex size back to 4 doesnt help.
Changing te3h PickRadius in Prefs doesnt help. 
Closing & opening the file ...
I opened the file in Ondsel & found the exact same behaviour. 
I did try FC 0.21, but that froze my box, so you can try that yourself. 

**Update**: I can, just, select some vertices by approaching from the right, usually top right, if nothing else is in the way.

My guess is that the drawing order is preventing selection. 
With larger verts, having selected one, I can see the shape of connecting wires on top 

My current use case is modelling a bicycle, which needs the ability to precisely join geometry in 3 dimensions with multiple directions, around the back wheel, so the ability to create & select exact locations becomes important. 

### Full version info

```shell
FC 1.0.0, but my guess is,"all of them".
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-03:
Please add an example file, maybe a screenshot or video of what the problem is. And add full about info. TIA

luzpaz, 2025-01-03:
If you don't know where it is, here's the wiki page [About](https://wiki.freecad.org/About) info :point_up:

gryphonB-2, 2025-01-04:
Version info:
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.5.10
  * Curves 0.6.51
  * DynamicData 2.70.0
  * slic3r-tools

gryphonB-2, 2025-01-04:
[BikeRack.FCStd.zip](https://github.com/user-attachments/files/18308514/BikeRack.FCStd.zip)
Uploaded file has simply been renamed with extension .zip & is in fact a FCStd file

gryphonB-2, 2025-01-04:
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.5.10
  * Curves 0.6.51
  * DynamicData 2.70.0
  * slic3r-tools

gryphonB-2, 2025-01-04:
**Update**
This is now a FIIK (Fd if I know). Having tried to reproduce the problem I can now select sketch verts easily from the main window, after I resize them to 8 pixels. That is, they act as expected. 
I can also resize verts on other sketches and select them easily from the main window without restarting. 

I would **request** though, that there be an entry in Preferences to allow general resizing of sketch vertices as seen from the main window, as is the case with Part/PartDesign, since in the default setup they are difficult to select.

gryphonB-2, 2025-01-05:
Final Update. 
While I stand by the above, there is a similar issue, also intermittent.
I'm finding that sketch wires are unselectable from certain viewpoints. 
This may be due to the background image, but problem persists with the image moved back 1mm from XZ plane
The wires become selectable when the viewpoint is suitably changed, which I hope the attached indicate.
[image]

gryphonB-2, 2025-01-05:
[image]

paranoidi, 2025-05-11:
[Unselectable.zip](https://github.com/user-attachments/files/20147162/Unselectable.zip)

Similar issue with 1.0, attached video of me trying to select lines and failing at max width. Includes project file in this exact situation for easy replication.

OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * ThreadProfile 1.94.0
  * fasteners 0.5.38

FEA-eng, 2025-11-18:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.20251104 (Git shallow)
Build date: 2025/11/04 19:07:44
Build type: Release
Branch: main
Hash: af014a1bb44264d749ba9522917f0d1ac76c4bbf
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```