FreeCAD/FreeCAD #18838: Failed to install the 64bit exe file on windows 10 64bit version.

yoyojacky, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
installed for several times, it stack on this error message .
OS: windows 10 22H2  19045.5247

### Full version info

```shell
FreeCAD 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-03:
@adrianinsaval what specific info is needed to track this down?

Syres916, 2025-01-03:
At that time in the install process, the files are being extracted to the TEMP location to the best of my knowledge, by default on the C: drive, so disk space and anti-virus interference are the first two places that I'd investigate.

closed 2025-10-04 by github-actions[bot]