FreeCAD/FreeCAD #18839: BIM: in "Meter decimal" units mode, height randomly changes value

furgo16, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Extract and open the test file with FreeCAD: [Patio roof_all_1.1.zip](https://github.com/user-attachments/files/18297472/Patio.roof_all_1.1.zip)
2. Select the `Structure` component in the document tree
3. On the Properties Editor, scroll down to make the `Height` property visible
4. Check the expected value (3800 mm) => OK
5. Go to the bottom right-hand corner of the FreeCAD window and click on the dropdown menu to select the `Meter decimal (m, m², m³)` unit system
6. Go back to the Properties editor, check the expected `Height` value (3800 m) => NOK 
7. Click on the `Height` value. Its value is increased automatically by 1000x (3800000 m) => NOK
8. Click on the `Height` value. Its value is increased automatically by 1000x (3800000000 m) => NOK
9. Click on the `Height` value again. Its value is now changed to a seemingly random number (-2147483648000 m) => NOK

See the attached screencast for a demo:

https://github.com/user-attachments/assets/4d577e60-62e3-482f-8a2a-73f57d7c1667

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Snap 1247
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
```

### Subproject(s) affected?

BIM

### Anything else?

This does not seem to be reproduceable in FreeCAD  1.0.0

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-03:
I have noticed the same but the result is the same as when I enter 10 and hit enter it assumes it is in mm despite the document being set to cm.

semhustej, 2025-01-04:
I can reproduce the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

I am not able to reproduce the issue with FC 1.0

semhustej, 2025-01-08:
It is not limited to BIM workbench. I encountered this issue in Draft WB. Steps to reproduce:

- open new file with draft workbench
- change unit system to `Meter decimal`
- start a rectangle command
- type 0 and hit enter as the X coordinate of the first point
- type 0 and hit enter as the Y coordinate of the first point
- type 0 and hit enter as the Z coordinate of the first point
- type 1 and hit enter as the X coordinate of the second point
- instead of 1m, 1000m is inserted
- type 1 and hit enter as the Y coordinate of the second point
- instead of 1m, 1000m is inserted
- select 0 and enter as the Z coordinate of the first point

See video:

https://github.com/user-attachments/assets/2e9b3178-60ba-4631-af9b-4252fba469a7

Roy-043, 2025-01-09:
It is indeed a regression. To pin-point it a bit further:

Bad:
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

Good:
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

Roy-043, 2025-01-11:
I think this is a Core issue.

semhustej, 2025-01-11:
> I think this is a Core issue.
> 
> 1. Unit system: `Standard`.
> 2. Create a new document.
> 3. Create a Part_Box (which is 10mm * 10mm * 10mm).
> 4. Change the unit system  of the document to `Meter decimal`.
> 5. The Property editor now says the box is 10m * 10m * 10m (it is not).
> 6. Click in the Length field.
> 7. Result: length jumps to 10000 m (and that is the actual length of the box then).

I am able to reproduce this issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39804 (Git) Conda
Build type: Release
Branch: main
Hash: 73f0e165bee7837704d374d0b1da430027e99fdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

furgo16, 2025-01-25:
@WandererFan you introduced the "Meter decimal" unit system back in the day (https://github.com/FreeCAD/FreeCAD/pull/11365). Do you have any ideas about what the cause of this issue could be? Thanks!

WandererFan, 2025-01-26:
It is impressive how much trouble a single wrong character can cause - PR #19259

closed 2025-01-28 by chennes