FreeCAD/FreeCAD #18842: Groove 2nd angle parameter not working

mccarthyam, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have created a Groove with two dimensions but changing the 2nd angle does not seem to have any impact.  Tried several combinations of angles and does not work. Encountered issue in Ver 1, downloaded the weekly build and did a quick test as per screenshot and file.

I need to specify the start and end points of the groove independent of the plane the sketch is on, my understanding is that 2nd angle should allow this 
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Conda
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * CADExchanger
  * Curves 0.6.47
  * ExtremeProDark 2.7.6
  * LCInterlocking
  * Lithophane
  * OpenTheme 2024.9.1
  * Render 2024.1.23
```

### Subproject(s) affected?

None

### Anything else?

[image]
[Groove problem.zip](https://github.com/user-attachments/files/18298795/Groove.problem.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-03:
Confirm

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.34; sheetmetal 0.6.13; Silk 0.1.5; Curves 0.6.53
```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.39748 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.34; sheetmetal 0.6.13; Silk 0.1.5; Curves 0.6.53
```

</details>

FEA-eng, 2025-07-03:
@maxwxyz Can it be somehow prioritized or considered a blocker? It's a pretty nasty regression.