FreeCAD/FreeCAD #18847: Hidden state of object's physical properies are not saved

1984-git, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Physical Properties (Material Name, Density etc) of a object can be hidden but after saving the document, restartarting FC and loading the document they appear again.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Europe (en_150)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2025-04-01:
Can you show steps to reproduce?

luzpaz, 2025-09-19:
ping @1984-git