FreeCAD/FreeCAD #18848: Slow loading IFC file

Isaina, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Loading IFC4_revit_Mep.ifc file need some time on free cad 1.0 and there is lots of processor power and memory free on apple m1 silicon

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-04:
Can you share the demo file?

Isaina, 2025-01-04:
Demo is fromhttps://github.com/youshengCode/IfcSampleFilesSent using the mail.com mail appOn 04.01.2025. at 11:28, Max Wilfinger wrote:
                
            
            
                
                    From: "Max Wilfinger" ***@***.***>Date: 4 January 2025To: "FreeCAD/FreeCAD" ***@***.***>Cc: "Author" ***@***.***>,"Isaina" ***@***.***>Subject: Re: [FreeCAD/FreeCAD] Slow loading IFC file  (Issue #18848)
                
                
Can you share the demo file?

semhustej, 2025-01-04:
I tried opening the _Ifc4_Revit_ARC.ifc_ sample file

With these settings it opens in 3min 09sec with 3.6GB RAM used
[image]

With these settings it opens in 29sec
[image]

In BonsaiBIM 0.8.0 the file opens in 27 seconds

Using old quad core notebook with Intel(R) Core(TM) i7-4710HQ CPU @ 2.50GHz

Isaina, 2025-01-04:
Yes also and I get faster loading on BonsaiBim so i creat this bug , is problem with settings only or there is a some bug in freecad

semhustej, 2025-01-04:
FreeCAD is doing a lot of work when opening/importing IFC files, that Bonsai does not do. This is why it takes longer time. I don't think this is a bug, but let's see what developers say.

Isaina, 2025-01-04:
Is there way to open some debug windows to see process which taking time

yorikvanhavre, 2025-01-06:
The import should be using multithreading. I'll have a look too

Isaina, 2025-01-17:
Is this related https://github.com/orgs/IfcOpenShell/projects/14/views/1?pane=issue&itemId=80638908&issue=IfcOpenShell%7CIfcOpenShell%7C5444

closed 2026-08-19 by Isaina