FreeCAD/FreeCAD #18849: sheetmetal unfold an a2plus

ik4yaz, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Problem
I generate an object with scheetmetal, u-shape,
I use unfold and it generates the flat sketch, it's ok, in techdraw I build the 2d table, it's ok.
but if I then want to use the object in a2plus it only inserts the flat and solid shape, not the "u".

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.68
  * dxf-library
  * fasteners 0.5.34
  * freecad.gears 1.3.0
  * Movie 2024.2.15
  * Render 2024.12.15
  * sheetmetal 0.6.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-03:
Wrong repo, either post to downstream to a2plus[^1] or Sheetmetal[^2]

[^1]: https://github.com/kbwbe/A2plus
[^2]: https://github.com/shaise/FreeCAD_SheetMetal

closed 2025-01-03 by luzpaz

ik4yaz, 2025-01-03:
sorry,sorry I'm not an expert, it's the first time I report a bug, now I follow your advice.
thanks.

luzpaz, 2025-01-03:
No worries. Thanks for making the effort to report. Hope this doesn't deter you from reporting in the future. We're just trying to keep the bugtracker as lean and clean as possible. :wink: