FreeCAD/FreeCAD #18850: Freecad1.0.0 crashs in TechDraw, Function Cosmetic_Circle

duebel273, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Dear Maintainer,
my congratulation for this very phantastic program, but you know it is never error free.
As a user I like to report an Error which appears on my Linux Mint 22 PC under FreeCad 1.0.0. , german version.

From my opened designed part I insert a A3_Landscape_ISO5457_advanced.svg sheet template.
I introduce a drawing using _TechDraw_View_ and mark the part.
I add a helping point using the menu item _TechDraw_CosmeticVertexGroup_ as the center point.
Then I add a helping point using the menu item _TechDraw_CosmeticVertexGroup_  as the radius point.
I mark both points and click on the symbol _TechDraw_Cosmetic_Cicle_.

After a short moment (apprx. 5 sec) Freecad crashes.

Thank you for your effort.
;-)
Peter

### Full version info

```shell
OS: KDE Flatpak runtime (MATE/mate/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-03:
Thanks for the report. Can you zip up the freecad file and then drag-drop it into the github textbox so it attaches to this ticket ?

WandererFan, 2025-01-04:
Just to be sure, you are using this cosmetic circle function:
[image]

and not this one:
[image]

WandererFan, 2025-01-05:
Can you remember anything else about what you were doing before the crash?   It does not crash here.

[image]

duebel273, 2025-01-05:
Hi,
sorry for the late response, here is the freecad-file which crashes:

[nU_Sonnenrad4c.zip](https://github.com/user-attachments/files/18311738/nU_Sonnenrad4c.zip)

duebel273, 2025-01-05:
@wandererfan:
I used the part design workbench to design my ring with 4 holes and saved it.
Then i change to the TechDraw workbench and insert a A3-Sheet.
Then I select my three lines part, pocket and circular pattern in the model three
and then I click on the symbol TechDraw_View to import my designed part.
I select the Front view and the Top view and click ok.
In the top view I select my ring by click on it and then I click on the symbol for TechDraw_CosmeticVertexGroup and accept the point position 0.0.
Then I select my center point and then one point on one of my 4 drills.
With this two selected points I click on the symbol for TechDraw_Cosmetic_Cicle.
After approx. 5 sec Freecad crashes.

greetings
duebel

duebel273, 2025-01-05:
@WandererFan :

sorry, I used add Cosmetic Circle, introduced in Version 1.0.
This crashes.

I realize now the existence of add Cosmetic Circle with three sub possibilities.
This one works also fine for me.

WandererFan, 2025-01-07:
I got a crash once, but I could not repeat it.  Does this happen for you every time you repeat the steps?

FEA-eng, 2025-09-20:
@duebel273 Can you retry in the newest weekly? I can't reproduce the issue (at least on Windows).

FEA-eng, 2025-11-21:
@WandererFan I still can't reproduce it, but it might be OS-specific. You've mentioned that it crashed for you once. Do you think this issue should stay open or rather be closed as not reproducible?

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251118 (Git shallow)
Build date: 2025/11/18 21:54:51
Build type: Release
Branch: main
Hash: be7d41dea75f7f2966d7b431d24f7060c6907339
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

FEA-eng, 2026-10-17:
> and also test once [#28635](https://github.com/FreeCAD/FreeCAD/pull/28635) passes review and merges ?
>      
>      
>      It's not merged yet, so maybe let's wait.

Still not merged.