FreeCAD/FreeCAD #18856: Core/FEM: Add CalcuiiX version info to About dialog

luzpaz, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

CalculiX is bundled with FreeCAD. To help with troubleshooting, the proposal is too add said version to the FreeCAD About info dialog. There is agreement from @marioalexis84 on this.

Propsal:
```
OS: Ubuntu Core 22 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
CalculiX 2.22
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Full version info

```shell
>=1.0
```

### Subproject(s) affected?

Core

### Anything else?

Discussion: 
* https://github.com/FreeCAD/FreeCAD/issues/18368#issuecomment-2543833788

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-01-04:
Maybe also Gmsh as it's included on Windows too (Linux builds have neither ccx nor gmsh included). It would also be good (and consistent) to include versions of other solvers (like Elmer) but I guess it would take too much space and be rarely needed. Perhaps there could be basic and extended/full About info?

luzpaz, 2025-01-04:
No, i agree, show all the solvers that are installed. 

I can't figure out how to find the solver versions using python

luzpaz, 2025-01-05:
Hi @oliveroxtoby, could you possibly weigh-in here? Since CfdOf has a very interesting solver installation process, I though maybe getting your perspective here would be valuable. I'm also wondering if there needs to be any conditional logic here for unorthodox installations (whatever those may be)?

oliveroxtoby, 2025-01-10:
Hi @luzpaz, no problem, although I'm not really sure how to help here? I guess just parsing the output of `ccx -version` would be the way to do it (using the path to calculix set up in the FEM options). Was there something specific you were after?

marioalexis84, 2025-01-10:
`ccx -v`

luzpaz, 2026-09-03:
soft bump