FreeCAD/FreeCAD #18858: Measurement::radius - Invalid References3D Provided  - No tools active just a face selected

pixelwaster, 2025-01-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I selected a face to use the Part_ColorPerFace and the above message "Measurement::radius - Invalid References3D Provided" popped up. I continued on with the color change. I selected another face and no message. The next two faces also had the message pop up. These were interior faces and don't have curves on them. The main shape is a box with fillets on the exterior.

Not the original file, but still gives Measurement::radius when selecting the curved face.

[Altoids 9V holder no switch.zip](https://github.com/user-attachments/files/18304217/Altoids.9V.holder.no.switch.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39747 (Git) Conda
Build type: Release
Branch: main
Hash: 7083362acb8529017b58ea03bf7416da0192b25b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * btl 0.9.9
  * CAM
  * CurvedShapes 1.0.13
  * Curves 0.6.53
  * Draft
  * Fem
  * freecad.gears 1.3.0
  * Help
  * Idf
  * Import
  * Inspection
  * lattice2 1.0.0
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * sheetmetal 0.6.13
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * toSketch 1.0.1
  * Tux
  * Web
```

### Subproject(s) affected?

Part

### Anything else?

My log file from the Altoids, it looks just like the original file. From main window to close, Is "Density is undefined" an issue to report?
```
Log: Init: Showing main window
Log: Main window restored
Log: Show main window
Log: Toolbars restored
Log: Loading PartDesign module... done
Log: Loading GUI of PartDesign module... done
Log: Init: Entering event loop
Log: Init: Executing event loop...
Log: Init: Processing command line files
Log: Hide main window
Log: Show main window
Log: Hide main window
Log: Show main window
Log: Loading TechDraw module... done
Log: Loading TechDrawGui module... done
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Sketcher::setUpSketch()-T:0.001687
Log: Sketcher::setUpSketch()-T:0.000672
Log: Density is undefined
Log: Density is undefined
Log: Sketcher::setUpSketch()-T:0.0007307
Log: Density is undefined
Log: Density is undefined
Err: Measurement::radius - Invalid References3D Provided
Err: Measurement::radius - Invalid References3D Provided
Log: Save AutoRecovery file in 0.086760s
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Log: Density is undefined
Err: Measurement::radius - Invalid References3D Provided
Log: Save AutoRecovery file in 0.042716s
Log: Hide main window
Log: Finish: Event loop left
Log: Destruct Gui::Application
Log: FreeCAD terminating...
Log: Saving system parameter...
Log: Saving system parameter...done
Log: Saving user parameter...
Log: Saving user parameter...done
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2025-01-04:
I've nog tested the sample file, but as another data point, I've started to see the `Measurement::radius - Invalid References3D Provided` message in one of my models. I've not managed to easily reproduce it, though.

Syres916, 2025-01-04:
This is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/18785

closed 2025-01-05 by maxwxyz