FreeCAD/FreeCAD #18861: Part Design: Pocket command gives a Pad

jlbirccyn, 2025-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

On the attached sketch, using Sketch011 to do a pocket result in a pad

[cles-22-7.FCStd.zip](https://github.com/user-attachments/files/18306264/cles-22-7.FCStd.zip)

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * A2plus 0.4.68
  * fasteners 0.5.33
  * LCInterlocking
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-04:
Confirm

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.34; sheetmetal 0.6.13; Silk 0.1.5; Curves 0.6.53
```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.39748 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.34; sheetmetal 0.6.13; Silk 0.1.5; Curves 0.6.53
```

</details>

tarman3, 2025-01-04:
Compaound [image]

Roy-043, 2025-01-04:
There are issues with several previous features. Some have faces with flipped normals. In fact all faces of the last feature have flipped normals. So the shape is inside-out.

See the 2nd note [here](https://wiki.freecad.org/Part_ReverseShape#Notes) on how to make this visible.

tarman3, 2025-01-04:
How does PartDesign create the surfaces with flipped normal?

Roy-043, 2025-01-04:
I have seen this before when fillets overlap. I would check those features first.

Beep6581, 2025-01-04:
I'm also experiencing the bug with a pocket resulting in a pad. See `PolarPattern001` in the attached file. `Sketch004 tab slits` is a pocket which cuts correctly, but the `PolarPattern001` of it looks like a XOR.

[cable spool 4.zip](https://github.com/user-attachments/files/18307205/cable.spool.4.zip)

[image]