FreeCAD/FreeCAD #18862: Std alignment not usable anymore

rmdug, 2025-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

When invoking std alignment tool from draft or BIM or part the alignment window will appear a very short time but FC comes back to the main window.
The alignment tab still appears, is selectable but FC always brings back the main view.

It is also not possible to close the alignment command. It is possible to close all documents but the alignment view remains empty and will not close until I restart FreeCAD.

So the std alignment tool is currently unusable for me.

Main view:
[image]

Aligment view (tricky to capture :)… ):
[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/Belgium (fr_BE)
Stylesheet/Theme/QtStyle: ProDark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Render 2024.12.15
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * Silk
  * FreeCAD-themes 2024.7.24
  * GDT
  * fasteners 0.5.34
  * fasteners.backup1735901309.314019 0.5.34 (Disabled)
  * Glass
  * InventorLoader 1.5.1
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[2025-01-04-01 - Essai alignement.zip](https://github.com/user-attachments/files/18306296/2025-01-04-01.-.Essai.alignement.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-04:
Is this also the case with the stable 1.0 version? Could you please try to restart FreeCAD in safe mode (Help>Restart in safe mode)

rmdug, 2025-01-04:
Hello maxwxyz.

I restarted in safe mode with weekly appimage and std alignment works normally.
[image]

With the official Manjaro release (stable ?):
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: French/Belgium (fr_BE)
Stylesheet/Theme/QtStyle: ProDark.qss/FreeCAD Dark/
Installed mods: 
  * Render 2024.12.15
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * Silk
  * FreeCAD-themes 2024.7.24
  * GDT
  * fasteners 0.5.34
  * fasteners.backup1735901309.314019 0.5.34 (Disabled)
  * Glass
  * InventorLoader 1.5.1

Here the situation is different: the problem exists but the alignment windows last just enough to let me select elements and click the align command to get it working, although not easily.

In safe mode with this version it works normally.

Thanks : )

maxwxyz, 2025-01-04:
So the issue is already resolved as it works with a fresh config file. You could try to delete your config files or play with the settings.

closed 2025-01-04 by maxwxyz

rmdug, 2025-01-07:
Hello,

After a few tuning on my new official FreeCAD the problem came back.
By disabling and enabling step by step the addons I identified that Glass might be the cause. Or should I say std alignment works well if Glass is disabled and won’t work as long as Glass is activated.

OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: French/Belgium (fr_BE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Render 2024.12.15
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * freecad.gears 1.3.0
  * Silk 0.1.5
  * FreeCAD-themes 2024.7.24
  * GDT
  * Glass (Disabled)

maxwxyz, 2025-01-08:
I guess the issue should be raised in the Glass repository if this creates the bug