FreeCAD/FreeCAD #18866: 'Execute Macro -> Toolbar' dialogue box size

semhustej, 2025-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Macro -> Macros... -> Toolbar -> `Do not show again` button is too small with unreadable text when using Light Theme:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2025-01-08:
@akym PR #18953 is a change to the core code but this is a _Stylesheet_ issue. If FreeCAD Classic or Behave-dark from https://github.com/FreeCAD/FreeCAD-themes are used then the button is perfectly fine:

[image]

[image]

There's a PR for stylesheet fixes in the queue, if I can find the easy tweak I'll raise it with the author.

akym, 2025-01-08:
Stylesheet solution is better, I will close the PR.

hyarion, 2025-03-01:
@Syres916 Do you know if the pr was merged with a fix for this?

Syres916, 2025-03-07:
Yes, #18929 definitely fixed this issue.

semhustej, 2025-03-07:
The issue seems to be fixed with current weekly. I am closing the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

closed 2025-03-07 by semhustej