FreeCAD/FreeCAD #18869: Solving the sketch failed - depends on zoom

Beep6581, 2025-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Solving the sketch seems to depend on the zoom level.

1. Part Design > Create Sketch.
2. Create circle by center, enter 1mm for the diameter.
3. Hit enter. Solving the sketch failed. Solver failed to converge.
4. Undo.
5. Zoom in with your mouse.
6. Create circle by center, enter 1mm for the diameter.
7. Hit enter. Works.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39757 (Git)
Build type: Release
Branch: main
Hash: a974e413fa39bbfa2572e99bc3af3824ad483263
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.34
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.13
  * Curves 0.6.53
```

### Subproject(s) affected?

None

### Anything else?

Screen recording:
[simplescreenrecorder-2025-01-04_19.03.36.zip](https://github.com/user-attachments/files/18307755/simplescreenrecorder-2025-01-04_19.03.36.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-01-04:
PS You can attach mp4 directly

Beep6581, 2025-01-04:
Thank you @tarman3 , I re-uploaded the mp4 directly.

For quick reference, e977eb49e3 is from 3 weeks ago, a974e413fa is from 23 hours ago.

[user.cfg.zip](https://github.com/user-attachments/files/18308557/user.cfg.zip)

maxwxyz, 2025-01-05:
Can you upload you FreeCAD file as ZIP?

tarman3, 2025-01-05:
> I re-uploaded the mp4 directly

Add empty line before and after video

Beep6581, 2025-01-05:
"Solver failed to converge": 
[18869.zip](https://github.com/user-attachments/files/18310475/18869.zip)

FlachyJoe, 2025-01-05:
I can reproduce here but it works as expected if auto-constraint is disabled.
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39757 (Git)
Build type: Debug
Branch: main
Hash: a974e413fa39bbfa2572e99bc3af3824ad483263
Python 3.12.8, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: French/France (fr_FR) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

Groberen, 2025-03-23:
Cannot reproduce in Version: 1.1.0dev.40655

Made  a forum thread containing further details, sample file and url to youtube:
https://forum.freecad.org/viewtopic.php?p=818011#p818011