FreeCAD/FreeCAD #18871: Physical Properties: Volume calculation not consistent

SebastianSchafer, 2025-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In some cases volume in the Part Design Body properties is not calculated properly. I have not been able to consistently reproduce it, but attached a file with 3 hollow cylinders of same dimensions all with different volume.
For the example of a hollow cylinder, the order of when the inner/outer circle in the sketch are created sometimes is causing a false calculation. The cylinders are 9mm (inner diameter) x 10mm (outer diameter) x 1m, resulting in a volume of 14.92ml (or cm<sup>3</sup>).
The bodies in the attached file have:
- 1: 93.46ml; this was confusing because the ratio is almost 2*pi - it turns out this is the volume of the 'full outer' cylinder (i.e., not hollowed out) plus the correct volume of the hollow cylinder (78.54ml + 14.92ml = 93.46ml). This was the initial body, with the outer circle created before the inner circle, followed by a pad.
- 2 & 3: These were created in the same matter as body1, though the volume is the correct 14.92ml
- 4: This body shows a negative volume of -48.69ml -- this is equivalent to the volume of the hollow cylinder minus the volume of the 'full inner' cylinder. Here, the inner circle was created before the outer circle.
[volume_debug.zip](https://github.com/user-attachments/files/18308073/volume_debug.zip)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.39745 (Git)
Build type: Release
Branch: main
Hash: 6bb424b5d16adffb1896187a0725945f9635d144
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * FreeCAD-themes 2024.7.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2025-09-20:
Why is this one assigned to me?

FEA-eng, 2026-02-20:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260217 (Git shallow)
Build date: 2026/02/17 21:26:14
Build type: Release
Branch: (HEAD detached at 3bad5dabd)
Hash: 3bad5dabde38f1a901fa1475e30f5caee7f9846a
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```