FreeCAD/FreeCAD #18872: Dimension text gets unreadable under zooming

carribeiro, 2025-01-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the sketcher, under some circumstances the dimension lines become unreadable depending upon the zoom level. I've narrowed down it considerably but it seems related to the use of external entities that are imported from other features or sketches.

I think it's weird but that's what I got after testing several combinations.

First, I designed a simple 20mm x 2mm rectangle in Safe Mode. It works ok.

Before zooming:
[image]

After zooming - text is the same size:
[image]

Now, I loaded another file where I noticed the problem first. After deleting a lot of unneeded features I got a simple version, where I have a slot which is positioned in relationship to an external reference - a circle drawn into another sketch. In this sketch the zoom breaks.

That's the view of the slot, zoomed out:
[image]

Now when I zoom in, it looks like this:

[image]

Text doesn't stay the same size actually - it gets SMALLER as I zoom in:

[image]

That's the entire FreeCAD window:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Conda
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

[ZOOMTEST.zip](https://github.com/user-attachments/files/18308130/ZOOMTEST.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2025-01-04:
I cannot reproduce this issue, text stays constant no matter the zoom. 

```
OS: Arch Linux (awesome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39757 +4 (Git)
Build type: Release
Branch: part-design-transparent-previews
Hash: a787fdc2a7ead7dffb20cf0dbb346bb09c5aeb63
Python 3.13.1, Qt 6.8.1, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Assembly3 0.12.0
  * SelectorToolbar
  * ExtremeProDark 2.9.0
  * CurvedShapes 1.0.4
  * IconThemes (Disabled)
  * fasteners 0.5.24
  * sheetmetal 0.3.0
  * OpenTheme 2024.9.1
  * Glass (Disabled)
  * Woodworking 0.21.2.33771
  * Gridfinity 0.5.0
  * lattice2 1.0.0
  * A2plus 0.4.60n
  * Curves 0.6.13
```

Syres916, 2025-01-04:
This appears to be a duplicate of #18516 , perspective seems to be the crucial setting.

carribeiro, 2025-01-04:
> I cannot reproduce this issue, text stays constant no matter the zoom.

Did you try with my test file? It works for me too if I use a file created from scratch.

kadet1090, 2025-01-04:
Nope, I tried with another file as it did not seem to be file related. Syres is right it seems to be related to perspecite as disabling it fixes the issues. I was also facing more issues with big zoom levels and perspective on new files.  I do confirm nowthat issue described by you is reproducible on your file.

maxwxyz, 2025-01-05:
Duplicate of #18516

closed 2025-01-05 by maxwxyz