FreeCAD/FreeCAD #18873: Editing a sketch causes View > Draw Style to be changed to 5 (No Shading) despite it showing ticked as 7 (Flat Lines).

gplsquared, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Editing a sketch causes View > Draw Style to be changed to option 5 (No Shading) despite it showing ticked as option 7 (Flat Lines). Clicking on option 7 puts objects rendered correctly in the scene. But having to keep doing it every time Sketcher is entered.

There is no reason for the UI to change draw style when a sketch is put into edit mode.

Thanks.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39748 (Git)
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.47
  * fasteners 0.5.29
  * kicadStepUpMod 11.3.2
  * Manipulator 1.5.7
  * Pyramids-and-Polyhedrons
  * freecad.gears 1.3.0
  * A2plus 0.4.67
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-05:
There is a preference to switch to no shading when entering the sketch. So the issue is, that the icon does not change?

gplsquared, 2025-01-05:
Thank you that was the issue. Once the 'feature' is disabled in sketcher preferences the draw style stops toggling when going back and forth from sketch edit to viewing objects in tree view. 
[image]

closed 2025-01-06 by maxwxyz