FreeCAD/FreeCAD #18874: PartDesign: Pockets in cylinders doesn't stop properly (to first/up to face)

Sam92-FC, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm doing (on FC 1.0) a modeling exercise suggested in this tutorial: [[https://youtu.be/F7ZnYQ7TQws?feature=shared](https://youtu.be/F7ZnYQ7TQws?feature=shared&fbclid=IwZXh0bgNhZW0CMTAAAR3JVZg9feG_YbyVzuoip82oKD-gQhQTacwTFYnfcs1Hn-lqQKpX3omKAT4_aem_Rk_ChoDqUIkBhUaIvuVMxQ)](https://www.youtube.com/watch?v=F7ZnYQ7TQws) (which dates from 2023).
Unfortunately, at the 16:04 mark of the video, things go differently for me.
When I want to do the leaning “pocket”, specifying “up to the first” to stop against the first vertical “pocket”, FC doesn't finish as properly as expected.
(I've tried different ways but the result isn't alright, sometimes FC does a kind of inverted Boolean operation...)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/Switzerland (fr_CH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

PartDesign

### Anything else?

**My .FCStd:**
[Base_reprise3.zip](https://github.com/user-attachments/files/18308960/Base_reprise3.zip)

**Screen recordings (2 versions of the problem)**

https://github.com/user-attachments/assets/50abb502-06d9-4c2e-9592-f057ddc1f7d8

https://github.com/user-attachments/assets/b480262a-c3d3-4e36-86d7-92e50d13c0e1

-----------------------------
As we can see in this tutorial from 2023 (at 16:04), it seems that, in a similar case on previous version of FC, function "Pocket" then "To first" was working: https://www.youtube.com/watch?v=F7ZnYQ7TQws

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-01-05:
No problem here
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39426 (Git)
Python 3.12.7, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
```
A full recompute of your file gives me the expected result.
[image]

The current main is compiling…

FlachyJoe, 2025-01-05:
I can not confirm with 
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39757 (Git)
Build type: Debug
Branch: main
Hash: a974e413fa39bbfa2572e99bc3af3824ad483263
Python 3.12.8, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: French/France (fr_FR) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

@Sam92-FC please retry with a recent version.

Sam92-FC, 2025-01-05:
_Sorry it’s the first time I post on Github and my understanding of English is "medium"._

What do you mean by "The current hand is compiling"?
When you say "retry with a recent version," are you referring to a recent version of Freecad?

**In fact, the expected result is as on the screenshot below: the vertical cylinder pocket goes straight down, and the tilted pocket should not go further than the vertical pocket.** 

[image]

What makes me think it’s a bug, is the fact that from time to time (**yet with the same file, which I put back in attachment), the reactions to Pocket are not the same** (see screen recording) ...then I tried to recalculate but it did not work.

In fact I tried to redo this model many times and always a bad result. I asked the question on a forum, and two experienced users confirmed that they have the same problem, either because of coplanities (even more with curves), or else they would get a different result if they placed the origin of the sketch elsewhere than in the center of the vertical cylinder (for example in an edge of the piece). The other person with the same problem (also on 1.1x dev build) advised me to use a plane datum to get around the problem.

[Ex.37_1 - Copiexxx.zip](https://github.com/user-attachments/files/18312132/Ex.37_1.-.Copiexxx.zip)

https://github.com/user-attachments/assets/aed117c9-01b1-4e48-b37e-c171413a5567

https://github.com/user-attachments/assets/8c49b6a2-242a-484e-916d-155745f1e610

[image]

FlachyJoe, 2025-01-06:
Ok got it, I thought the reported bug was the remaining solid in the vertical cylinder bottom, this shape disappear in the recent FreeCAD versions I tested.
This issue problem is the pocket going through all instead of stopping at the vertical cylinder. It works well if the _Sketch001_ edge intersect the cylinder face but not if it goes below. So it's a face extension problem. I'll take a look in the code soon.

Workaround: put the pocket sketch in the vertical cylinder and do the pocket in the other direction.

[FR]
Ok j'ai compris, je pensais que le bug était qu'il restait un morceau de solide en bas du cylindre vertical. Ce morceau disparaît avec les versions récentes de FreeCAD que j'ai testé.
Le problème rapporté ici est donc que la poche va à travers tout au lieu de s’arrêter sur le cylindre vertical. Ça fonctionne comme il faut si la ligne de l'esquisse _Sketch001_ intersecte la face du cylindre mais pas si elle passe dessous. C'est donc un problème d'extension de la face. Je regarderai dans le code d'ici peu.

Proposition de contournement : décale l'esquisse de la poche pour qu'elle soit dans le cylindre vertical et fais la poche dans l'autre sens.

_"The current main is compiling"_ signifiait que j'étais en train de compiler le code le plus à jour (la version de développement courante) sur mon ordinateur. Son test a donné le rapport que j'ai fait dans mon second message.
[/FR]

Sam92-FC, 2025-01-06:
Yes, that's exactly it!  

_And "merci" for the explanations and tips :)_

FlachyJoe, 2025-01-15:
With the minimal example below I can find a threshold on the transversal cylinder diameter: for a 5mm vertical cylinder the cut fails at a 4.6136mm transversal diameter but works at 4.6135mm. And this doesn't depend on the angle.
[image]
[ToFace_Bug.FCStd.zip](https://github.com/user-attachments/files/18430202/ToFace_Bug.FCStd.zip)

FEA-eng, 2025-09-20:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2026-10-16:
> This issue problem is the pocket going through all instead of stopping at the vertical cylinder. It works well if the Sketch001 edge intersect the cylinder face but not if it goes below. So it's a face extension problem. I'll take a look in the code soon.

@FlachyJoe any chance you can follow-up here ?