FreeCAD/FreeCAD #18877: Clipping UI needs improving

greggjaskiewicz, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Clipping Plane (View->Clipping Plane) is a very useful feature of FreeCAD, however it seems it suffers from quite a few usability issues. 
- The UI is not resizable , meaning that its hard to fit it with rest of the UI
- The UI persists, even tho the use of it suggests that it's something user should be able to toggle without having to see the panel on the screen whilst it's toggled on
- It does not persist parameters, meaning that each time you dismiss it - and you need to enable it again, you have to set everything up from scratch. It should be persisted per project, or even allow user to have multiple pre-defined ones. 

Source of frustration? I am designing a box, and would like to clip side of it , two different sides in fact to be able to work on features inside the box. However, use of it lead to all the mentioned frustrations. 

This might be an easy starter feature to implement. See my use of it for reference
[image]

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39741 (Git) Conda
Build type: Release
Branch: main
Hash: 0d128517dedd154bbecfbabe67094ad56c8d46f5
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
  * AnimationFreeCAD 1.0.0
  * A2plus 0.4.68
  * sheetmetal 0.6.2
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.52
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-06:
IMO a completely new implementation would be better, see https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue+is%3Aopen+label%3A%22Topic%3A+Clipping+Planes%22+

IIRC @shaise was working on clipping planes

luzpaz, 2026-10-28:
Related: 
* #8477

Faehnchen, 2026-10-29:
I'll link here a Reddit post from yesterday where the bad behaviour of Clipping View is documented: https://www.reddit.com/r/FreeCAD/s/LXmhOhdZl4

OP described how the clipping view was pushing their lower window bar of FreeCAD out of the screen boundary and made it at the moment impossible for them to go back to the normal window aspect. u/R2W1E9 could reproduce the behaviour and provided a way out of the situation (recorded with a screencapture): [Link](https://www.reddit.com/r/FreeCAD/comments/1xhj9lq/comment/19rrzqzbo99a8/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)

I myself couldn't reproduce the error in  Version: 27.1.0dev.20261027 AppImage and Version: 1.1.4.45039 Flatpak., but found that the Clipping View dialog when opened or dragged in a floating panel in the viewport, obstructs the close-button. Only when the panel border is dragged way far to the right, the close-button appears at the furthest right corner.
This behaviour is different from most other panels like the Measurements- or Sketcher panel, where the close-button is always centered and visible at the top or bottom of the panel. I documented it with a short screencapture as well: [Link](https://www.reddit.com/r/FreeCAD/comments/1xhj9lq/comment/19rs5zhhsz9l1/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)

From a user perspective this potentially very useful tool should be an integral part of FreeCAD and probably be overhauled. Right now it provides minimal functionality, is not well integrated into the graphical UI and the open cross section faces do not provide clarity and are not nice to look at. The macro [FreeCAD-LiveSectionAnalyzer ](https://github.com/sedatkrn/FreeCAD-LiveSectionAnalyzer) already provides in many aspects superior and innovative features and something like that should imo be something to show off and  therefore be part of the FreeCAD core package.