FreeCAD/FreeCAD #18879: PartDesign Additive Pipe does not correctly handle self-intersection

Simonski-GitHub, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When the result of a sweep is self-intersecting, the PartDesign AdditivePipe operation does not detect this and creates an invalid solid, i.e. edges are not created at places where parts of the sweep overlap, or removed at places within overlaps. See screenshot for an example of sweeping a profile along an intersecting path.

[image]

The resulting sweep can be manually checked for validity by running Check Geometry with the 'Run Boolean operation check' flag set so a quick and dirty half-fix would be to add this check after the sweep; if the result contains self-intersections, produce an explanatory warning message to the user.

However, Additive Pipe should generate the correct topology in the first place.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.34
  * OpenTheme 2024.9.1
  * sheetmetal 0.6.13
  * Assembly4 0.50.15
  * Render 2024.12.15
  * Help 1.0.3
  * freecad.gears 1.3.0
  * Curves 0.6.53
```

### Subproject(s) affected?

PartDesign

### Anything else?

See https://forum.freecad.org/viewtopic.php?p=801693#p801693 for some discussion.

Possibly related issues:
https://github.com/FreeCAD/FreeCAD/issues/18658 
https://forum.freecad.org/viewtopic.php?t=58298

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-06:
Maybe related:
- https://github.com/FreeCAD/FreeCAD/issues/15589
- https://github.com/FreeCAD/FreeCAD/issues/17705

ElementW, 2026-08-25:
Still self-intersecting as of faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and either OCC 7.9.3 or 8.1.0.dev1-3d097a0328