FreeCAD/FreeCAD #18881: TechDraw: title block font rendering

pmjdebruijn, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When exporting PDFs the text in the title block becomes outlined (as opposed remaining searchable text), and I haven't been able to see why this is happening. This doesn't happen with SVG exports though.

I noticed this because the title block text (using the ISO templates) renders particularly poorly (looks bolded with zoomed out, but looks proper when zoomed in).

When I change PDF/A output to PDF 1.6 output, the poor text rendering goes away (with the text still remaining outlined, thus unsearchable).

```
sed -i 's#QPagedPaintDevice::PdfVersion_A1b#QPagedPaintDevice::PdfVersion_1_6#g' src/Mod/TechDraw/Gui/PagePrinter.cpp
sed -i 's#QPagedPaintDevice::PdfVersion_A1b#QPagedPaintDevice::PdfVersion_1_6#g' src/Mod/Spreadsheet/Gui/SpreadsheetView.cpp
sed -i 's#QPagedPaintDevice::PdfVersion_A1b#QPagedPaintDevice::PdfVersion_1_6#g' src/Gui/MDIView.cpp
sed -i 's#QPagedPaintDevice::PdfVersion_A1b#QPagedPaintDevice::PdfVersion_1_6#g' src/Gui/EditorView.cpp
sed -i 's#QPagedPaintDevice::PdfVersion_A1b#QPagedPaintDevice::PdfVersion_1_6#g' src/Gui/ApplicationPy.cpp
sed -i 's#QPagedPaintDevice::PdfVersion_A1b#QPagedPaintDevice::PdfVersion_1_6#g' src/Gui/View3DInventor.cpp
```

I'm not sure sure why this has such a big impact on text legibility.

### Full version info

```shell
1.0.0+41~gd9e8fe0d7d with PDF/A patch backported
Qt 5.15.8 (on Linux)
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pmjdebruijn, 2025-01-05:
[Unnamed-Page.pdf](https://github.com/user-attachments/files/18310962/Unnamed-Page.pdf)

WandererFan, 2025-01-05:
IIRC, in v1.0, printing from the context menu produced a PDF1.4 file, and printing from the file menu produced a PDF/A-1b file (PDF1.4 with additional restrictions).  see #18443 for background.

Do you get a different result with these two methods?

pmjdebruijn, 2025-01-05:
I've actually dug through some old files I had, and the plain PDF1.4 doesn't seem to have the issue either... So PDF -1.6 doesn't fix it, not using PDF/A fixes it.

pmjdebruijn, 2025-01-05:
As a note, in 1.0 PDF version was inconsistent, you got PDF/A in some instances, and PDF-1.4 in others.

WandererFan, 2025-01-06:
I don't understand the "outlined" comment.  It just looks like bold text to me.

[image]

pmjdebruijn, 2025-01-06:
Sorry for the confusion, I meant outlined as in vectorized... 

Which practically means the text can be selected anymore, thus not copy-pasted.

And that also search document management systems, won't be able to index/keyword search these PDFs very well either.

mpfusion, 2025-01-07:
I can reproduce the rendering issue here. It's vector data in both cases and the issue more of less vanishes when zoomed in. But when zoomed out the newer FreeCAD produces quite bold text.
Current:
[image]
Before:
[image]

pmjdebruijn, 2025-01-07:
Basically, the aliasing? (perceived boldness) is the immediate practical issue...

However, if the text to vector conversion didn't happen this issue probably wouldn't have occurred in the first, which is why I brought it up. So if that's something can be changed, that would be the preferable solution.

tobiasfalk, 2025-01-08:
The Text to Vector conversion is somthing that QT does when importing a SVG, realy annoying. FreeCAD would need to grap the Text elements and redraw them the same way it draws all the other Text.
I have posted about Text problems already a while ago:
https://forum.freecad.org/viewtopic.php?t=69791
https://forum.freecad.org/viewtopic.php?t=78177

pmjdebruijn, 2025-02-19:
Until the root cause of this issue is found and solved, can some please do sometimes like this at least on `releases/FreeCAD-1-0` ?

```
for F in $(grep -Hr 'PdfVersion_A1b' * | awk -F ':' '{print $1}'); do sed -i 's,[A-Za-z]*\.setPdfVersion.*;,#&,g' "${F}"; done 
```

3nt3, 2025-04-06:
My current workaround is to just export as svg and convert to pdf manually, the text is selectable then. Just need to make sure you have osifont (or whatever font) installed on your system.

alerque, 2025-05-19:
The recent #21096 seems to be an entire duplicate of this. In spite of the initial misdiagnosis there is some info there that might be relevant though: namely that the version of FreeCAD may actually *not* directly affect this and the same version of FreeCAD when compiled against newer versions of dependencies exhibits this problem. That info might help narrow down what is going on.

I can try to provide some context about what library versions were current at the time of the new/old builds if info about a relevant library is requested.

tobiasfalk, 2025-05-21:
Ok, this may sound weared, but it could be a PDF version of export problem.

See #20860 and #18921 , 
Yes, it is definitely not the same issue, but both affect the PDF export.

luzpaz, 2025-05-27:
@tobiasfalk Thank you for making that upstream ticket!

tobiasfalk, 2025-05-27:
I made it a year ago because I had problems my self but did not thought to inform the FreeCAD community about it

tobiasfalk, 2025-05-28:
Also I am not 100% sure if this QT Issue is related to this FreeCad Issue.

But I know that I posted about the path export problem in the forum 3 years ago.
https://forum.freecad.org/viewtopic.php?t=69791&sid=92d264c3eb05955c3548742c562a5615

tobiasfalk, 2025-06-03:
The Issue was closed after the apparent fix was merged.

I have no experience in compiling QT and then use it to compile FreeCAD, so I would ask if someone could test it with the newest qr dev build

alerque, 2025-06-03:
I don't have time to rebuild Qt, then run it for the sake of rebuilding this to test. Sorry, I just don't have time right now.

If you wanted to do so and since you happen to be on Arch, the procedure would probably be use your favorite AUR helper to build and install `*-git` variants of all QT related dependencies, then rebuild the official freecad package. It is possible somebody has a user repository with QT development packages available too that could save a lot of building, but I don't know who that might be. Look in the Arch Wiki at the user repo list.

As soon as QT has a release and Arch packages it, FreeCAD will get rebuilt against it anyway as a matter of routine. At that point it should be obvious if that was the fix or not.

alerque, 2025-06-19:
Thanks @tvrusso that is immensely useful information.

In that case it looks like the bug here is the upstream Qt6 issue [QTBUG-123817](https://bugreports.qt.io/browse/QTBUG-123817) and it is marked as fixed in the upcoming 6.9.2. That will land in Arch pretty much as soon as it is released.

furgo16, 2025-11-23:
The output, as shown in https://github.com/FreeCAD/FreeCAD/issues/18881#issuecomment-2575420969 is quite poor. I just attempted to print some drawings and I can confirm.

> When I change PDF/A output to PDF 1.6 output, the poor text rendering goes away (with the text still remaining outlined, thus unsearchable).

Could we not do this as a compromise for FreeCAD 1.1, so that at least the text is legible? It will take quite a while until Qt 6.9.1, which reportedly fixes the bug, makes it into FreeCAD.

WandererFan, 2025-11-23:
The appearance issue appears to be fixed in Qt 6.8.3.  I don't know about the searchable text issue.

This is the export from recent weekly AppImage:
[image]

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.2.0dev.20251119 (Git shallow) AppImage
Build date: 2025/11/19 23:28:17
Build type: Release
Branch: main
Hash: deab0abf8e61d69f2208b794f64ac56428467fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/118.229/1.19792
Installed mods: 
  * A2plus 0.4.68
  * sheetmetal 0.7.58
  * fasteners 0.5.43
```

WandererFan, 2025-11-24:
@furgo16 you can set your pdf version here:

[image]

luzpaz, 2025-12-25:
@WandererFan looks like the PDF version changes all reflect the problem.

FEA-eng, 2026-08-21:
Indeed the text is not selectable/searchable

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260804 (Git shallow)
Build date: 2026/08/04 21:26:11
Build type: Release
Branch: (HEAD detached at 80923ab1b)
Hash: 80923ab1be1f31fff0a54db956177d582a658bab
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

WandererFan, 2026-08-21:
Qt 6.8.3

AFAIK, this is fixed on Win in builds using Qt 6.9.3+.  It has been fixed on linux for a while.