FreeCAD/FreeCAD #18882: Unintended switch between Model and Tasks in Treeview when switching between windows (to / from Spreadsheet)

M4xTheMan, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Start**

- Treeview is configured to have seperate tabs for Model and Tasks (Tree view and Property view mode: Combined)
[image]

- Two windows are open (PartDesign and Spreadsheet)
- The PartDesign window is selected. In the Treeview, the tab Model is selected.
[image]

**expected behavior**
When switching between the Spreadsheet window and the PartDesign window, in the Treeview the tab Model should stay selected.

**current behavior**
Switching back from the Spreadsheet window to the PartDesign window leads to the selection of the tab Tasks in the Treeview.
[image]

[image]

**additional information**
This does not happen when switching between a PartDesign window and a TechDraw window.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39748 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 11.4.4
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.3
  * Manipulator
  * fasteners 0.5.34
  * Assembly3 0.12.2
  * sheetmetal 0.6.13
  * InventorLoader 1.5.1
  * Assembly4
  * Curves 0.6.53
  * POV-Ray-Rendering
  * OpenDark 2024.3.13
  * 3DfindIT 1.2.0
  * Gridfinity 0.6.0
  * Freecad-Built-in-themes-beta 1.2.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2025-01-09:
Tools menu -> edit parameters -> baseapp -> preferences -> mod -> part design -> create new boolean "SwitchToTask" set to False.  That should prevent the switching.

M4xTheMan, 2025-01-10:
Thank you @mwganson , I'll try that. I didn't realize that this seems to be the default behavior of the PartDesign WB. It looked to me like it would depend on the WB I'm coming from (hence the Spreadsheet WB / TechDraw WB description in my initial post).

I did a little research and found this thread in the FC forum from 09.2023: https://forum.freecad.org/viewtopic.php?t=80923
Furthermore, there's a related issue with a feature request to make the parameter "SwitchToTask" available via the preferences: https://github.com/FreeCAD/FreeCAD/issues/13322

@maxwxyz : What do you suggest how to handle this issue?

1. Should we close it as intended behavior (= me misunderstanding the intended behavior of the PartDesign WB)?
2. Should this be addressed by the Design Working Group (= should this be the default or not?)?
 
I prefer Nr. 2

maxwxyz, 2025-01-11:
@FreeCAD/design-working-group what is your opinion?

M4xTheMan, 2025-02-22:
@FreeCAD/design-working-group ping

MisterMakerNL, 2025-02-22:
The program should not switch any panels out of it self without communicating it to the user.
Now the issue here is I never use combined task window, so I am not a user of this system, so it easy to make this statement I do not know what I am talking about. 
I am just saying Freecad isn't communicating what it is doing which can confuse the user.

FEA-eng, 2025-09-20:
@M4xTheMan Can you retry with the newest weekly? Now the SwitchToTask parameter is available as a preference. But even with the setting allowing the switch, I can't reproduce the issue.

M4xTheMan, 2025-11-28:
When I enable the PartDesign setting SwitchToTask, it's switching to Task and stays there. I'd expect that it would go back to model when leaving PD. Has this been discussed somewhere?

**Regarding the behaviour I've reported here**
I can't reproduce it with the weekly version below (SwitchToTask not activated). Therefore it's now working as expected. For the record: I've switched to another distro (Ubuntu -> Fedora).

Thanks! 

I'm going to close this issue. If anyone knows if the SwitchToTask behaviour has been discussed somewhere, I'd be happy to get a pointer in the right direction.

`OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.2.0dev.20251119 (Git shallow) AppImage
Build date: 2025/11/19 23:28:17
Build type: Release
Branch: main
Hash: deab0abf8e61d69f2208b794f64ac56428467fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/127.472/2
Installed mods: 
  * Freecad-Built-in-themes-beta 1.2.2
  * Gridfinity 0.11.9
  * InventorLoader 1.5.1
  * LCInterlocking
  * Lithophane
  * OpenDark 2024.3.13
  * POV-Ray-Rendering
  * Curves 0.6.70
  * Defeaturing 1.2.9
  * fasteners 0.5.43
  * FusedFilamentDesign 0.25.380
  * Manipulator 1.6.3
  * sheetmetal 0.7.58
  * Assembly3 0.12.3
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.9
`

closed 2025-11-28 by M4xTheMan