FreeCAD/FreeCAD #18885: Dropdown arrow disappears when hovering mouse over features

Condyloma, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The dropdown arrow disappears when the mouse hovers over a specific tool or feature. Although the arrow disappears, the dropdown menu can still be used by clicking the spot where the arrow should be. This behavior may cause confusion due to the lack of visual guidance.

**Steps to Reproduce**
1.    Open FreeCAD.
2.    Locate any tool in the toolbar that has a dropdown arrow for additional tools.
3.    Hover the mouse over the tool.
4.    Notice that the dropdown arrow disappears when interacting with the feature. However, clicking the location where the arrow should be still activates the dropdown menu.

[image]

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Core

### Anything else?

Suggestion for Usability Improvement

To improve the user experience, the dropdown menu could automatically open when the mouse hovers over the tool. This enhancement would eliminate the need for the arrow entirely, streamlining the workflow.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

InRiPa, 2025-01-06:
I think this is the same issue as found here:
https://github.com/FreeCAD/FreeCAD/issues/17982

Condyloma, 2025-01-06:
> I think this is the same issue as found here: #17982

Thank you for pointing out the issue "Wayland: Display errors and main window resizing." However, after reviewing the description, I believe my issue is different.

In my case:

    The dropdown arrow disappears when I hover over a tool with a dropdown menu, but the menu itself is still functional. Clicking the spot where the arrow should be opens the menu as expected.
    This issue seems to be a visual inconsistency rather than a functional failure.

In the Wayland issue, the dropdown menus are completely hidden and non-functional, which doesn’t match my experience.

Syres916, 2025-01-06:
@Condyloma it's a stylesheet specific issue which hopefully won't take too much tracking down

MisterMakerNL, 2025-01-07:
ai I think you are right @Syres916 I'll make a patch together with some other things!

closed 2025-01-27 by yorikvanhavre