FreeCAD/FreeCAD #18887: CAM: Tool Library / Tool Bits Inconsistent use of instructions across OSs

JohnOCFII, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I thought there was already a Github issue for this, but could not find it.  I know there was discussion on the FreeCAD Discord about this issue.

When creating the initial ToolBits Library in the CAM workbench, and during other processes in the CAM workbench regarding ToolBits and/or the Tool Library, there is instructional text placed in the operating system's dialog boxes (at least on Linux) that do not appear on Macintosh.  Ideally these instructional dialog boxes would be created in a way that is consistent across all supported operating systems.

I've included one example between macOS and Debian Linux to show the issue.

[image]

[image]

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39748 (Git) Conda
Build type: Release
Branch: main
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2025-05-26:
@knipknap - Since you are looking at Tool Library changes, I wanted to see if this might fit in your skill set or interest area.

knipknap, 2025-05-26:
I think this is a duplicate of #18887. But honestly I am not sure that I understand this correctly; is the expectation that all OS should have the same type of window title? I think file dialogs exist precisely to avoid apps from doing that, to provide a more native experience for each platform. Its a conscious choice made by each OS.

PR #21425 (2025-06-02): CAM: Integration of the tool and shape editors from BTL (Better Tool Library)
This very large PR essentially replaces CAM Tool Management completely, as part of the [Better Tool Library](https://github.com/knipknap/better-tool-library/) integration.

## Relevant Issues

Discussion: #9466

I also went through all open CAM issues to find any that may be resolved or otherwise affected:

- Resolves: #8201
- Resolves: #8526
- Resolves: #9625
- Resolves: #11522
- Resolves: #12712
- Likely resolves: #12823, but should be tested to be sure
- Possibly resolves #13603, to be tested
- Resolves: #17083
-  [... cut]