FreeCAD/FreeCAD #18889: Core: Drag/Dropping .FCstd file into GUI doesn't trigger recomputation dialog (where opening the file from within FC does)

luzpaz, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Scenario 1**: Drag-drop an FCStd file from my file manager into to FreeCAD  
Result: it open the file uneventfully.

**Scenario 2**: File > Open  
Result: 'Recomputation Required' dialog is shown.

To reproduce:  
1. Download the example file from https://github.com/FreeCAD/FreeCAD/issues/9512#issuecomment-1545023153  
2. Try Scenario 1
3. Close file
4. Try Scenario 2

### Full version info

```shell
I can reproduce this on both the development builds in both Snap and AppImage on Manjaro Linux

OS: Ubuntu Core 22 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39757 (Git) Snap 1251
Build type: Release
Branch: main
Hash: a974e413fa39bbfa2572e99bc3af3824ad483263
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-06:
@wwmayer do you have a moment to diagnose this?

Syres916, 2025-01-08:
@luzpaz I may have a PR later this evening (it works but needs some tidying yet) but @wwmayer could no doubt code it more succinctly.

luzpaz, 2025-01-22:
This really needs attention. It may seem like a non-blocker but this is a problem when triaging. If going through the effort of testing a bug can be skewed by a faulty workflow, it wastes everyone's time.

Syres916, 2025-01-23:
@luzpaz I'll resubmit a new PR based on the feedback, it's the bit from @tritao about moving into Document that I need to thoroughly test and the fact that the Migration Check has been removed in the days while it's been in the queue, I'll do my best to have it ready tonight UK time.

Syres916, 2025-01-24:
@luzpaz I've tried my best to explain the tests carried out and the code layout in #19211 , hopefully it's acceptable or at least needs very minimal changes.

PR #19211 (2025-03-05): [Gui] Trigger the same checks for Drag & Dropping files as File > Open
Replaces https://github.com/FreeCAD/FreeCAD/pull/18951

@tritao regarding previous feedback, I can confirm testing a 0.21.x file there is a single `checkForRecomputes` triggered for both Drag & Drop and File > Open
Also tested a version 1.0 file to ensure that there wasn't a Recompute triggered, as well as a Step and a DXF file, all worked as I would expect.

I've left the checks in Application primarily because I tried and failed to compile it any other way. With Werner creating them as separate checks I'm not inclined to  [... cut]
closed 2025-03-05 by chennes