FreeCAD/FreeCAD #18890: Frequent SEGV crash in Assembly when selecting features to create joints

BjoKaSH, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD (on Linux)  crashes randomly, i.e. without obvious pattern, when selecting edges or vertices for subsequent joining by a (in my case, probably irrelevant) fixed joint.  I have now seen five(!) crashes within about two hours of work on a fairly simple assembly.  If memory serves right, three of the five crashes where while working on a sub-assembly, the other on the main assembly (only involving part directly added to the main assembly, not involving any sub-assembly).

The file contains only two bodies, both referenced from a second file using a sub-shape binder.  Several copies of each of the two bodies are added to the assembly using the toolbar icon to add parts, so these are added as links (if this is the right term, I am new to FreeCAD and its terminology).

I checked existing issues #18705, #17162 and #15202, but they seem to be different.
However, I am also affect by the #18044 problem: After the crash restarting FreeCAD produces many "```Reading failed from embedded file: <some_file_name>.bin (0 bytes, 2 bytes compressed)```" messages.  Luckily, only the recovery file is corrupted in my case, and the manually saved files are still fine.

All crashes produced the same stack-trace in the console window where FreeCAD was started:

```Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x75a3db442520]
#1  0x75a3c32f7066 in MbD::Part::qX(std::shared_ptr<MbD::FullColumn<double> >) from /tmp/.mount_FreeCALbdGoo/usr/lib/./libOndselSolver.so.1+0x6
#2  0x75a3c31c9e68 in MbD::ASMTSpatialContainer::updateMbDFromPosition3D(std::shared_ptr<MbD::FullColumn<double> >) from /tmp/.mount_FreeCALbdGoo/usr/lib/./libOndselSolver.so.1+0xb8
#3  0x75a3c4527a67 in Assembly::AssemblyObject::doDragStep() from /tmp/.mount_FreeCALbdGoo/usr/lib/AssemblyApp.so+0x1e7
#4  0x75a3c4324ced in AssemblyGui::ViewProviderAssembly::tryMouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /tmp/.mount_FreeCALbdGoo/usr/lib/AssemblyGui.so+0x35d
#5  0x75a3c43256ec in AssemblyGui::ViewProviderAssembly::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /tmp/.mount_FreeCALbdGoo/usr/lib/AssemblyGui.so+0x1c
#6  0x75a3debb46a0 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x400
#7  0x75a3daed346e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0xbe
#8  0x75a3dacd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x68
#9  0x75a3dae8cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0xea
#10  0x75a3daedc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x58
#11  0x75a3daef713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x2a
#12  0x75a3dacd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x68
#13  0x75a3dae8cb9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0xea
#14  0x75a3daedc348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x58
#15  0x75a3daef713a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x2a
#16  0x75a3dacd54e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x68
#17  0x75a3dacdf26c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0xac
#18  0x75a3dacd61ba in SoAction::apply(SoNode*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x2ea
#19  0x75a3daea8ad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libCoin.so.80+0x72
#20  0x75a3deb3714e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x1e
#21  0x75a3deb42618 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x58
#22  0x75a3deb4c360 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x40
#23  0x75a3deb51cbb in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x2db
#24  0x75a3deb4e265 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x65
#25  0x75a3deb77c3b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x6b
#26  0x75a3deb2d9e5 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x1a5
#27  0x75a3dbab68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Core.so.5+0x7b
#28  0x75a3dc97039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#29  0x75a3dc9778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#30  0x75a3de769559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0xc9
#31  0x75a3dbab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Core.so.5+0x108
#32  0x75a3dc9763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#33  /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x75a3dc9c6e6f]
#34  /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x75a3dc9ca0c9]
#35  0x75a3dc9703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#36  0x75a3de769559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0xc9
#37  0x75a3dbab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Core.so.5+0x108
#38  0x75a3dc1353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Gui.so.5+0x84c
#39  0x75a3dc10b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Gui.so.5+0xac
#40  /tmp/.mount_FreeCALbdGoo/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x75a3d62d5bea]
#41  /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x75a3da110ad7]
#42  /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x75a3da113ac7]
#43  /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x75a3da1140ce]
#44  0x75a3dbb0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Core.so.5+0x66
#45  0x75a3dbab570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Core.so.5+0x11b
#46  0x75a3dbabd212 in QCoreApplication::exec() from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libQt5Core.so.5+0x82
#47  0x75a3de6c8744 in Gui::Application::runApplication() from /tmp/.mount_FreeCALbdGoo/usr/bin/../lib/libFreeCADGui.so+0x8a4
#48  /tmp/.mount_FreeCALbdGoo/usr/bin/freecad(+0x78b7) [0x632ef58d48b7]
#49  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x75a3db429d90]
#50  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x75a3db429e40]
#51  /tmp/.mount_FreeCALbdGoo/usr/bin/freecad(+0x7c2a) [0x632ef58d4c2a]
```

### Full version info

```shell
OS: TUXEDO OS 3 (LXDE/LXDE/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

BjoKaSH, 2025-01-05:
Short Update: Next crash, same stack-trace, this time selecting a face.  The crash occurred right after the face turned dark green after clicking it.  It was the second face selected, while holding Ctrl key pressed to add as second selected feature.

Sequence of actions was: Select a body in tree view.  Zoom and rotate using mouse wheel and middle button to see its bottom face,  select that face (left click, it turned light green).  Zoom out a bit using mouse wheel.  Point mouse to other nearby body in 3D view, zoom, pan and rotate with mouse wheel and middle button to see its top face.  Left-click top face. Crash.

Note: 3D Navigation mode is set to "Blender".

derlucas, 2025-01-06:
I got a related issue with the Assembly Workbench.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7ff9e684c1d0]
#1  0x7ff9e6f5305b in QCoreApplicationPrivate::lockThreadPostEventList(QObject*) from /usr/lib/libQt6Core.so.6+0x6b
#2  0x7ff9e6faa0f4 in QObject::deleteLater() from /usr/lib/libQt6Core.so.6+0x34
#3  0x7ff9e7972c5e in QBasicDrag::eventFilter(QObject*, QEvent*) from /usr/lib/libQt6Gui.so.6+0x38e
#4  0x7ff9e1671eeb in QtWaylandClient::QWaylandDrag::finishDrag() from /usr/lib/qt6/plugins/platforms/../../../libQt6WaylandClient.so.6+0xbb
#5  /usr/lib/qt6/plugins/platforms/../../../libQt6WaylandClient.so.6(+0xb41ce) [0x7ff9e16671ce]
#6  /usr/lib/libffi.so.8(+0x7596) [0x7ff9e21f1596]
#7  /usr/lib/libffi.so.8(+0x400e) [0x7ff9e21ee00e]
#8  /usr/lib/libffi.so.8(ffi_call+0x123) [0x7ff9e21f0bd3]
#9  /usr/lib/qt6/plugins/platforms/../../../libwayland-client.so.0(+0x48b0) [0x7ff9e15a88b0]
#10  /usr/lib/qt6/plugins/platforms/../../../libwayland-client.so.0(+0x5139) [0x7ff9e15a9139]
#11  /usr/lib/qt6/plugins/platforms/../../../libwayland-client.so.0(wl_display_dispatch_queue_pending+0x73) [0x7ff9e15a9553]
#12  0x7ff9e1613426 in QtWaylandClient::QWaylandDisplay::flushRequests() from /usr/lib/qt6/plugins/platforms/../../../libQt6WaylandClient.so.6+0x66
#13  0x7ff9e6fa2d9a in QObject::event(QEvent*) from /usr/lib/libQt6Core.so.6+0x15a
#14  0x7ff9e7efe78a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8a
#15  0x7ff9ea0a7289 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#16  0x7ff9e6f57ac8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x1a8
#17  0x7ff9e6f58555 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt6Core.so.6+0x195
#18  /usr/lib/libQt6Core.so.6(+0x3c4f7c) [0x7ff9e71c4f7c]
#19  /usr/lib/libglib-2.0.so.0(+0x5d559) [0x7ff9e570d559]
#20  /usr/lib/libglib-2.0.so.0(+0xc0257) [0x7ff9e5770257]
#21  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7ff9e570ca55]
#22  0x7ff9e71c22fd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#23  0x7ff9e6f61ee6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x226
#24  0x7ff9e796f4ab in QBasicDrag::drag(QDrag*) from /usr/lib/libQt6Gui.so.6+0x4b
#25  0x7ff9e796fe6d in QDragManager::drag(QDrag*) from /usr/lib/libQt6Gui.so.6+0xfd
#26  0x7ff9e79701cd in QDrag::exec(QFlags<Qt::DropAction>, Qt::DropAction) from /usr/lib/libQt6Gui.so.6+0x8d
#27  0x7ff9e8203ebc in QAbstractItemView::startDrag(QFlags<Qt::DropAction>) from /usr/lib/libQt6Widgets.so.6+0x19c
#28  0x7ff9ea39c3c3 in Gui::TreeWidget::startDrag(QFlags<Qt::DropAction>) from /usr/lib/freecad/lib/libFreeCADGui.so+0x23
#29  /usr/lib/libQt6Widgets.so.6(+0x400eec) [0x7ff9e8200eec]
#30  0x7ff9e81f54e7 in QAbstractItemView::mouseMoveEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0x517
#31  0x7ff9e7f4f51f in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x97f
#32  0x7ff9e7fee138 in QFrame::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0x38
#33  0x7ff9e6f54a08 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x118
#34  0x7ff9e7efe77a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x7a
#35  0x7ff9e7f02a26 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x1076
#36  0x7ff9ea0a7289 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#37  0x7ff9e6f57ac8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x1a8
#38  0x7ff9e7ef613d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x21d
#39  /usr/lib/libQt6Widgets.so.6(+0x169752) [0x7ff9e7f69752]
#40  /usr/lib/libQt6Widgets.so.6(+0x16ae40) [0x7ff9e7f6ae40]
#41  0x7ff9e7efe78a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8a
#42  0x7ff9ea0a7289 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xd9
#43  0x7ff9e6f57ac8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x1a8
#44  0x7ff9e758752b in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x3db
#45  0x7ff9e7602b44 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x314
#46  /usr/lib/libQt6Gui.so.6(+0x64ea14) [0x7ff9e7a4ea14]
#47  /usr/lib/libglib-2.0.so.0(+0x5d559) [0x7ff9e570d559]
#48  /usr/lib/libglib-2.0.so.0(+0xc0257) [0x7ff9e5770257]
#49  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7ff9e570ca55]
#50  0x7ff9e71c22fd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#51  0x7ff9e6f61ee6 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x226
#52  0x7ff9e6f58f4f in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x9f
#53  0x7ff9e9faf3b0 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xcf0
#54  freecad(+0x6cd4) [0x559b08d0bcd4]
[... cut ...]
#56  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7ff9e6834ecc]
#57  freecad(+0x70d5) [0x559b08d0c0d5]
```

My system is:
```
Linux hostname 6.6.65-1-MANJARO #1 SMP PREEMPT_DYNAMIC Wed Dec 11 22:24:04 UTC 2024 x86_64 GNU/Linux
FreeCAD 1.0.0 Revision: 39109 (Git)
```

Some time this evening i even got the destroyed Project file like mentioned in https://github.com/FreeCAD/FreeCAD/issues/18044

The Error occured when i tried to move a Body in the Assembly a couple of items in the tree up (sorting Bodies).